russhughes/st7789_mpy #2: Rotation issue

aleggon, 2020-06-19:
Hi.
I am using this excellent driver with a 240x240 display, just like the one on the picture. Everything works good with rotation 0 and 1. With rotation 2 and 3 the display seems to be offset.
What may be the problem?

russhughes, 2020-06-19:
The memory in the controller chip configured for a 240x320 display.  When using a smaller display like a 240x240 or 135x240 an offset is added to the x and y parameters so that the pixels are written to the  memory area that corresponds to the visible display.   When you rotate the display the offsets change need to change.  The only display I've used is the 135x240. This display uses x+53, y+40 for rotation 2,  and x+40, y+52 for rotation 3. 

I think the 240x240 display will need x+0, y+80 for rotation 2, and x+79, y+0 for rotation 3. I don't have a 240x240 display to test this, but I can add the ability to set the offset values and you can see what numbers work for your display. If you let me know what values work for you I can update the driver with them.

russhughes, 2020-06-19:
Here is a link to the new firmware you can try: [https://penfold.owt.com/esp32/firmware.bin](https://penfold.owt.com/esp32/firmware.bin).  After you call **rotation** you can call **offset(x, y)** to set the x and y offsets for your display.  I think these numbers should work.

```
tft.rotation(2)
tft.offset(0, 80)
```

or

```
tft.rotation(3)
tft.offset(79, 0)
```

aleggon, 2020-06-20:
Hi. Following your instructions I changed the source code to put the offsets for rotations 2 (x=0, y=80) and 3 (x=80, y=0). I also had to change it to restore the zero offsets (x=, y=0) for the case of returning to rotations 0 and 1 (see the attached st7789.c). Now it`s working like a charm. Thank you very much!
[st7789.c.zip](https://github.com/russhughes/st7789_mpy/files/4807478/st7789.c.zip)

russhughes, 2020-06-20:
Excellent, I'll update the repo later today.  Thanks for your help.

russhughes, 2020-06-20:
Added new offset method, fixed the offsets for 240x240 displays and updated the firmware.bin

closed 2020-06-20 by russhughes