russhughes/st7789_mpy #3: Pyboard error

aleggon, 2020-06-27:
Hi. Sorry to bother you again. I admit that I'm not more than a begginer Arduino C++ programmer.
I tried to compile the C module for Pyboard v1.1 and got these warnings/errors:

../../../st7789_mpy/st7789/st7789.c:423:26: error: implicit declaration of function 'malloc' [-Werror=implicit-function-declaration]
  423 |     uint16_t *c_buffer = malloc(buf_size);
      |                          ^~~~~~
../../../st7789_mpy/st7789/st7789.c:423:26: error: incompatible implicit declaration of built-in function 'malloc' [-Werror]
../../../st7789_mpy/st7789/st7789.c:34:1: note: include '<stdlib.h>' or provide a declaration of 'malloc'
   33 | #include "st7789.h"
  +++ |+#include <stdlib.h>
   34 | //#include <stdlib.h> //coloquei para stm32
../../../st7789_mpy/st7789/st7789.c:455:9: error: implicit declaration of function 'free' [-Werror=implicit-function-declaration]
  455 |         free(c_buffer);
      |         ^~~~
../../../st7789_mpy/st7789/st7789.c:455:9: error: incompatible implicit declaration of built-in function 'free' [-Werror]
../../../st7789_mpy/st7789/st7789.c:455:9: note: include '<stdlib.h>' or provide a declaration of 'free'
cc1: all warnings being treated as errors
../../py/mkrules.mk:63: recipe for target 'build-PYBV11/st7789/st7789.o' failed
make: *** [build-PYBV11/st7789/st7789.o] Error 1

As suggested, I added #include <stdlib.h> to the code but it was not enough to fix it.
Is the module intended for use with ESP32 only?

russhughes, 2020-06-27:
I've only used it with an ESP32 but I do have a Pyboard v1.1 that I haven't used yet, let me see what I can figure out.

russhughes, 2020-06-27:
Changing from malloc to alloca to allocate the character buffer 'c_buffer' seems to work.  The buffer could be changed to a static array large enough to hold the largest sized character you plan to use (width * height * 2) and it would probably be faster at the cost of about 1K of RAM.  I haven't tested this update with an actual display but it compiles and imports on my pyboard1.1. I'll dig out a display and do some testing later today.

russhughes, 2020-06-27:
I have verified that the fix works on my pyboard 1.1 and 240x240 tft.

closed 2020-06-27 by russhughes