russhughes/st7789_mpy #4: is the firmware working?

jhfoo, 2020-07-07:
i have a 2020 t-watch. looking to use your firmware to get ST7789V working!

russhughes, 2020-07-07:
The current firmware files in the repo will probably not work on the T-Watch.  The T-Watch has 8MB PSRAM and a 16MB flash,  it is different then the T-Display.  I can compile a version that might work on the T-Watch if you would like to test it.

russhughes, 2020-07-08:
Try the firmware in the esp32-spiram-16flash directory.

jhfoo, 2020-07-08:
Wow thank you! Trying now.

jhfoo, 2020-07-08:
I'm not sure it's to do with your build: was able to at least put some pixels on the [slow devbis library](https://github.com/devbis/st7789py_mpy) last night. Today nothing worked on the same code, even on fresh hardware. Gotta fix that before I can verify your firmware I'm afraid.

If it's any help here's my unsuccessful test code on your firmware:
```
from machine import SPI, Pin
import st7789
import vga2_bold_16x32 as font
import random

spi = SPI(2, baudrate=30000000, polarity=1, phase=1, sck=Pin(18), mosi=Pin(19))
display = st7789.ST7789(
    spi,
    240, 240,
    reset=Pin(23, Pin.OUT),
    cs=Pin(5, Pin.OUT),
    dc=Pin(27, Pin.OUT),
    backlight=Pin(12, Pin.OUT),
    rotation=0)
display.init()
display.fill(st7789.color565(5,5,5))
display.text(font, 'KungFOO rocks!', 50,50,
    st7789.color565(
        random.getrandbits(8),
        random.getrandbits(8),
        random.getrandbits(8)),
    st7789.color565(
        random.getrandbits(8),
        random.getrandbits(8),
        random.getrandbits(8))
    )
```

And here's the [official pinout](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/src/board/twatch2020_v1.h).

jhfoo, 2020-07-08:
Though it was listed as a supported language, online contacts at Lilygo offer support only on C/ C++. This is misleading and frustrating.

I still think this is a wonderful developer watch. You may be my best chance to get this in a working condition. Would you be interested to receive a watch of the same model in return for providing working firmware/ code for the screen?

russhughes, 2020-07-08:
I have a watch on order that I'll be able to use for testing once it arrives.  There is no reset pin listed for the display, pin 23 is used for the TOUCH_SDA, that parameter should be removed. 

I'll dig into the documentation and Arduino library and see what I can find. Did the firmware work as far as showing you a REPL prompt?  Did the file system work? Were issues limited to just the display?

jhfoo, 2020-07-08:
> I have a watch on order that I'll be able to use for testing once it arrives. There is no reset pin listed for the display, pin 23 is used for the TOUCH_SDA, that parameter should be removed.
>
> I'll [...]

Yes the firmware booted up fine, and my AXP202 code worked as expected. I was able to manage files with Thonny and VS Code (pymakr).

Only spent time on the display; I can validate wifi after this.

jhfoo, 2020-07-11:
Sharing a success repo someone posted on the micropython forum for the same hw:
https://gitlab.com/mooond/t-watch2020-esp32-with-micropython

Ref: https://forum.micropython.org/viewtopic.php?f=18&t=8642&p=49059#p49059

russhughes, 2020-07-24:
My watch arrived today so I'll start working on it.

russhughes, 2020-07-24:
I have a preliminary build working with the watch and should have something to release this weekend.

russhughes, 2020-07-25:
Updated firmware and new example program for TTGO T-Watch-2020.

closed 2020-07-25 by russhughes

jhfoo, 2020-07-25:
Left kudos for your work on the FB ESP32 group: https://www.facebook.com/groups/2773813706042809