russhughes/st7789_mpy #5: BMA423 support

jhfoo, 2020-07-08:
Creating a new issue to avoid contaminating the last one: will you know how to get BMA423 supported on micropython? LoBo included in his firmware built, but that's outdated already. 

If there is a way I can learn and help, please point me in the right direction. :) This is also for the 2020 T-watch.

russhughes, 2020-07-16:
I’ll look into it when my watch arrives.

jhfoo, 2020-07-25:
Gentle callout for this support. 😄

jschrbr, 2020-07-25:
https://github.com/lewisxhe/MicroPython_ESP32_psRAM_LoBo

also here 

https://github.com/OPHoperHPO/lilygo-ttgo-twatch-2020-micropython

jhfoo, 2020-07-25:
> https://github.com/lewisxhe/MicroPython_ESP32_psRAM_LoBo

Not convinced this firmware works for the 2020 version. Drivers listed appear to be for the 2019 model (ie no ST7789S).
> 
> also here
> 
> https://github.com/OPHoperHPO/lilygo-ttgo-twatch-2020-micropython
Looks super complete! Wonder how it got past my earlier googling results.

jhfoo, 2020-07-25:
Closing unless the author(s) are planning to tackle this in-house.

jhfoo, 2020-08-09:
A little embarassed to ask, but is there any way you can embed the BMA423 driver in your firmware? If there is an existing micropython library I would use it.

russhughes, 2020-08-09:
Bluetooth, bma432, ir, st7789, axp202c and frozen vga fonts in firmware3.bin from the firmware/ttgo_watch folder.

closed 2020-08-09 by russhughes

jhfoo, 2020-08-10:
OMG you're such a gem!