russhughes/st7789_mpy #6: ImportError: no module named 'axp202c'

jhfoo, 2020-07-25:
Uploaded [firmware](https://github.com/russhughes/st7789_mpy/blob/master/firmware/ttgo_watch/firmware.bin) into the TT-go watch, and ran the [example script](https://github.com/russhughes/st7789_mpy/blob/master/examples/watch_hello.py) by renaming as boot.py.

Watch booted up with the above message. Detailed capture below
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5020
load:0x40078000,len:11608
load:0x40080400,len:6116
entry 0x400806bc
I (639) psram: This chip is ESP32-D0WD
I (639) spiram: Found 64MBit SPI RAM device
I (639) spiram: SPI RAM mode: flash 40m sram 40m
I (642) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (649) cpu_start: Pro cpu up.
I (653) cpu_start: Application information:
I (658) cpu_start: Compile time:     Jul 24 2020 19:19:43
I (664) cpu_start: ELF file SHA256:  0000000000000000...
I (670) cpu_start: ESP-IDF:          v3.3.2
I (675) cpu_start: Starting app cpu, entry point is 0x40084320
I (0) cpu_start: App cpu up.
I (1551) spiram: SPI SRAM memory test OK
I (1551) heap_init: Initializing. RAM available for dynamic allocation:
I (1551) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1557) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1564) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1570) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (1576) heap_init: At 3FFCAB80 len 00015480 (85 KiB): DRAM
I (1582) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1589) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1595) heap_init: At 4009A4F4 len 00005B0C (22 KiB): IRAM
I (1601) cpu_start: Pro cpu start user code
I (278) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Traceback (most recent call last):
  File "boot.py", line 12, in <module>
ImportError: no module named 'axp202c'
MicroPython v1.12-464-gcae77daf0-dirty on 2020-07-24; ESP32 module (spiram) with ESP32
Type "help()" for more information.
```

jschrbr, 2020-07-25:
make sure you have [axp202c.py](https://github.com/russhughes/st7789_mpy/blob/master/lib/axp202c.py) in the same directory as boot.py

or in boot.py add the axp202c.py directory to system paths

`import sys
sys.path[1] = '/path/to/dir'`

jhfoo, 2020-07-25:
Thanks for clarifying! I thought I read that axp202c is baked into the firmware. Should've dug into the lib/ folder. 

Watch demo works, and is awesome! Thanks much! Closing ticket.

Aside: will you be exploring BLE on the watch?

closed 2020-07-25 by jhfoo

russhughes, 2020-07-25:
It should be in the firmware but I may have missed something. I’ll check on it.