russhughes/st7789_mpy #8: Document building instructions

jhfoo, 2020-07-28:
Hi team. I suspect the steps you took to build this firmware is not all as documented in the README.MD; can you confirm?

Will you be able to identify the steps? I'm a happy user of your firmware, and thought to swap in/ out some of the fonts. Perhaps if we put our heads together we can automate an auto build of the firmware, similar to https://nodemcu-build.com/? I have a Pi3 that can be used to do the autobuild.

Happy to know your thoughts.

russhughes, 2020-07-31:
Once you have all the build tools running building the firmware is as easy as described in the README.  For the Watch I modified the partitions.csv file for the larger flash memory and copied the python files to be frozen into the modules directory.  I'll look into what it would take to make an online build tool.

jhfoo, 2020-07-31:
The steps are quite a bit; my first attempt failed. Will try and re-read the steps.

robhamerling, 2021-02-21:
Tried to build the firmware today for the ESP32, following probably an earlier version of the README.md. No compiler errors, but firmware doesn't contain the st7789 module. 
Now I see in README a slightly different make command including: `-DMODULE_ST7789_ENABLED=1`
This gives a make error with me, the -D option seems not supported (Linux Mint 20.1). What can I do?

russhughes, 2021-02-21:
Try this: 

make CFLAGS_EXTRA="-DMODULE_ST7789_ENABLED=1" USER_C_MODULES="../../../st7789_mpy" all

If the module is not in the firmware the source couldn't be found.  When running make, one of the first lines printed should say "Including User C Module from ../../../st7789_mpy/st7789"   with a line father down saying: "CC ../../../st7789_mpy/st7789/st7789.c"  showing that it was actually compiled.

robhamerling, 2021-02-22:
Used this make command literally (copy-paste), but still no st7789 module. And I do not see the expected CC line. 
When I do a  'cd ../../../st7789_mpy' I really arrive in the st7789_mpy directory, so the st7789 code should be found (at least I find it in the st7789 subdirectory!).

russhughes, 2021-02-22:
I will check this again later today. I usually use a shell script to compile several different versions of the firmware at once so I may have missed something.

russhughes, 2021-02-23:
That command line works for me when compiling the MicroPython 1.14 source from https://micropython.org/resources/source/micropython-1.14.tar.xz using ESP IDF v4.  If you are using the latest source from github it probably wont work as it has been change to use CMake. See https://forum.micropython.org/viewtopic.php?f=18&t=9820&p=55249&hilit=cmake#p54986 for more information.  You can use the github source if you use the 1.14 tag.

robhamerling, 2021-02-23:
OK! I saw indeed the move to cmake, but wasn't aware of the consequences w.r.t. the st7789 module.
Your 1.14 version in the firmware directory works fine, but it was indeed my intention to compile the latest 'unstable' 1.14 source. Hopefully you'll be able to convert your setup to cmake  soon.

russhughes, 2021-02-23:
I plan to look at CMake this weekend.

closed 2021-02-23 by russhughes

picpic020960, 2021-03-15:
Bonjour Russ ,
some news ?
Thank's