russhughes/st7789_mpy #9: A smoother text drawing option

jhfoo, 2020-07-28:
Sharing my breaking success integrating your work with a prior library used for SD1306 screen (CWriter).

I note that refreshing the screen for text changes involves noticeable screen blinking as the screen goes black. This is due to the st7789.mpy implementation of writing directly to the screen.

The library includes the blit_buffer() method, which copies a FrameBuffer object basically a bytearray to the screen. I tested using The CWriter class to paint the text into the buffer, and update the screen with blit_buffer(). That seems to stop the blinking artifact.

Pretty excited over this success, and thought to share the approach with you. Happy to update my repo with sample code if you're interested.

jhfoo, 2020-07-29:
Rendering 10 lines of text using CWriter requires about 4sec to paint into the buffer, though the refresh experience is fantastic.

I think the best option is to update the original st7789_mpy C code to write into a bytearray buffer.

russhughes, 2020-07-31:
That wouldn't be hard to do.  I used the direct write method since the device I was using has limited RAM memory. Since the watch has more RAM it might be better to work on a framebuffer compatible version.

jhfoo, 2020-07-31:
Sharing my ST7789 Framebuffer class that is tested to interop with CWriter
```
# std
import framebuf

# 3rd party
from writer import CWriter

# custom 
import font6

SCR_WIDTH = 240
SCR_HEIGHT = 240

class ST7789Buffer(framebuf.FrameBuffer):
    def __init__(self):
        self.width = SCR_WIDTH
        self.height = SCR_HEIGHT
        self.buffer = bytearray(self.height * self.width * 2)
        super().__init__(self.buffer, self.width, self.height, framebuf.RGB565)
        self.font = font6
        self.writer = CWriter(self, self.font, 0xffff, 0)

    def renderText(self, text, x, y, fgcolor=0xffff, bgcolor=0x0000):
        CWriter.set_textpos(self, y, x)
        self.writer.setcolor(fgcolor, bgcolor)
        self.writer.printstring(text)

    def show(self, driver):
        driver.blit_buffer(self, 0, 0, SCR_WIDTH, SCR_HEIGHT)
```

closed 2021-06-02 by russhughes