russhughes/st7789_mpy #12: BMA423 sample code

jhfoo, 2020-08-10:
**What I did**
I used the sample code from [here](https://github.com/lewisxhe/MicroPython_ESP32_psRAM_LoBo), modifying slightly to the following code:
```
import bma423

i2c =I2C(scl=Pin(32), sda=Pin(23))
bma423.init(i2c)
```

SCL and SDA pins were selected, following the callout [here](https://github.com/Xinyuan-LilyGO/TTGO_TWatch_Library/blob/master/docs/pinmap.md) (quote):  _SENSOR SDA and SCL share PCF8563, BMA423, AXP192,T-Block without BMA423, use MPU6050_

**OUTPUT**
OS rebooted.

**The ask**
Can you review the actual driver code and confirm with an example how the driver should be used?

russhughes, 2020-08-11:
I see the issue. The driver uses a different i2c library.  I'll have to modify the driver.

jhfoo, 2020-08-11:
That makes sense, seeing the stock example asks for Pin number instead of the Pin object. Thanks for looking into this!

jhfoo, 2020-08-18:
Hi @russhughes. Wondering if you had any success on this?

russhughes, 2020-08-18:
Not yet.

mtasatcom, 2020-09-21:
Hi, I was wondering if you incorporated the bma423 driver into the ttgo-t-watch build.

russhughes, 2020-09-21:
I'm still working in it.

russhughes, 2021-06-02:
I'm no longer working on it.

closed 2021-06-02 by russhughes