russhughes/st7789_mpy #13: map_bitarray_to_rgb565 use

cr0mbly, 2020-08-13:
Hi there are you able to provide an example of the `map_bitarray_to_rgb565` I want to generate icons for display on the watch that have a coloured forground and transparent background, I've tried to convert the png image into a bytearray and pass it to this method and display by blitting but to no luck, any help would be appreciated :pray:

russhughes, 2020-08-14:
I'll come up with one this weekend.

russhughes, 2020-08-17:
I was able to work around an odd invalid parameter type issue by changing the interface code. It works now and I'll have new firmware with example code uploaded later tonight.  I think I'm going add the ability to specify the number of bits per pixel in the bitarray to support indexed colored bitmaps.

russhughes, 2020-08-18:
Uploaded driver code, updated T-Watch firmware and new example watch_bitarray.py.