russhughes/st7789_mpy #14: Core dump on initialization

Ktrod, 2020-08-20:
I cross compiled using the modified `pycom-esp-idf` and am getting a `Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.` on initialization of the st7789. 

I've included the current thread at core dump. Perhaps an issue with Pycom's cross compiler?

`===============================================================
==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
pc             0x40210b9b	0x40210b9b <mp_virtual_pin_write+15>
lbeg           0x40093c90	1074347152
lend           0x40093c9b	1074347163
lcount         0x0	0
sar            0x1c	28
ps             0x60e20	396832
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x40115056	1074876502
a1             0x3ffd8200	1073578496
a2             0x3ffc0330	1073480496
a3             0x0	0
a4             0x1	1
a5             0x1	1
a6             0x3f414a04	1061243396
a7             0x3ffd8210	1073578512
a8             0x0	0
a9             0xa	10
a10            0x3ffc0330	1073480496
a11            0x2	2
a12            0x0	0
a13            0x0	0
a14            0x3	3
a15            0x60023	393251

==================== CURRENT THREAD STACK =====================
#0  0x40210b9b in mp_virtual_pin_write (pin=0x3ffc0330 <pin_GPIO14>, value=0) at ../extmod/virtpin.c:38
#1  0x40115056 in st7789_ST7789_hard_reset (self_in=0x3f94ffb0) at mods/modst7789.c:173
#2  st7789_ST7789_init (self_in=0x3f94ffb0) at mods/modst7789.c:386
#3  0x40100838 in fun_builtin_1_call (self_in=0x3f414bb0 <st7789_ST7789_init_obj>, n_args=1, n_kw=0, args=0x3f94ef18) at ../py/objfun.c:70
#4  0x400fd0b0 in mp_call_function_n_kw (fun_in=0x3f414bb0 <st7789_ST7789_init_obj>, n_args=1, n_kw=0, args=0x3f94ef18) at ../py/runtime.c:624
#5  0x400fd134 in mp_call_method_n_kw (n_args=0, n_kw=0, args=0x3f94ef14) at ../py/runtime.c:640
#6  0x401085ee in mp_execute_bytecode (code_state=0x3f94ef00, inject_exc=0x0) at ../py/vm.c:1002
#7  0x4010093b in fun_bc_call (self_in=<optimized out>, n_args=<optimized out>, n_kw=0, args=0x0) at ../py/objfun.c:287
#8  0x400fd0b0 in mp_call_function_n_kw (fun_in=0x3f94ee50, n_args=0, n_kw=0, args=0x0) at ../py/runtime.c:624
#9  0x400fd0dd in mp_call_function_0 (fun=0x3f94ee50) at ../py/runtime.c:598
#10 0x400e146d in parse_compile_execute (source=0x3f94ed66, input_kind=MP_PARSE_FILE_INPUT, exec_flags=32) at ../lib/utils/pyexec.c:103
#11 0x400e16f0 in pyexec_file (filename=0x3f94ed66 <error: Cannot access memory at address 0x3f94ed66>) at ../lib/utils/pyexec.c:560
#12 0x400e01aa in TASK_Micropython (pvParameters=<optimized out>) at mptask.c:320`

russhughes, 2020-08-20:
I don't have any experience with the pycom-esp-idf and don't know if there are any differences in how it handles pin I/O.  It looks like something went wrong in the st7789_ST7789_hard_reset routine were it toggles the pin used to reset the ST7789 controller.  

What device are you running this on?

Ktrod, 2020-08-21:
Im running a FiPy right now. Pycom creates pin objects a bit differently than the standard micropython variant so that might be an issue but as both are still pin objects I couldn't see why.

russhughes, 2020-08-21:
That makes sense.  I'll do a little reading to see if I can make it compatible with their hardware abstraction layer.

Ktrod, 2020-08-28:
Ok I fixed the issue with writing to the GPIO pins by removing the macro calls to `mphal.h` and using pycoms variant of writing to the pins.

```
#define mp_hal_pin_write(p, v) do { \
		pin_obj_t *npin = pin_find(p); \
		npin->value = v; \
		pin_set_value(npin); \
	} while (0)

#define mp_hal_pin_read(p) pin_get_value(pin_find(p))
```

However I am now getting a kernel panic when attempting to write to the SPI. This is using the provided example code and the incident occurs when attempting the transfer.

The following is the core dump

```
================== CURRENT THREAD REGISTERS ===================
pc             0x4021e84b	0x4021e84b <write_spi+15>
lbeg           0x40093b78	1074346872
lend           0x40093b83	1074346883
lcount         0x0	0
sar            0xb	11
ps             0x60b20	396064
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x40117455	1074885717
a1             0x3ffd9c20	1073585184
a2             0x3ffcb558	1073526104
a3             0x3ffd9c40	1073585216
a4             0x1	1
a5             0x1	1
a6             0x16f	367
a7             0x13f7	5111
a8             0x0	0
a9             0x3ff4400c	1072971788
a10            0x3ffcb558	1073526104
a11            0x1	1
a12            0x3ffd9c40	1073585216
a13            0x0	0
a14            0x0	0
a15            0x0	0

==================== CURRENT THREAD STACK =====================
#0  0x4021e84b in write_spi (spi_obj=0x3ffcb558 <mach_spi_obj>, buf=0x3ffd9c40 \"\\001\	\\020\\200p\\234\\375?\\220\\002\\225?\", len=1) at mods/modst7789.c:56
#1  0x40117455 in write_cmd (self=0x3f950290, cmd=1 '\\001', data=0x40117455 <write_cmd+77> \"\\246\\025'\\242\\\"\\006\\245\\231\315\202\\032\
1\\261\\336\\060\\070 2Z\
\\300 \", len=1073585184) at mods/modst7789.c:97
#2  0x40117b7c in st7789_ST7789_soft_reset (self_in=0x3f950290) at mods/modst7789.c:198
#3  st7789_ST7789_init (self_in=0x3f950290) at mods/modst7789.c:606
#4  0x4010095c in fun_builtin_1_call (self_in=0x3f42a7e4 <st7789_ST7789_init_obj>, n_args=1, n_kw=0, args=0x3f94ef38) at ../py/objfun.c:70
#5  0x400fd1d4 in mp_call_function_n_kw (fun_in=0x3f42a7e4 <st7789_ST7789_init_obj>, n_args=1, n_kw=0, args=0x3f94ef38) at ../py/runtime.c:624
#6  0x400fd258 in mp_call_method_n_kw (n_args=0, n_kw=0, args=0x3f94ef34) at ../py/runtime.c:640
#7  0x4010871a in mp_execute_bytecode (code_state=0x3f94ef20, inject_exc=0x0) at ../py/vm.c:1002
#8  0x40100a5f in fun_bc_call (self_in=<optimized out>, n_args=<optimized out>, n_kw=0, args=0x0) at ../py/objfun.c:287
#9  0x400fd1d4 in mp_call_function_n_kw (fun_in=0x3f94ee70, n_args=0, n_kw=0, args=0x0) at ../py/runtime.c:624
#10 0x400fd201 in mp_call_function_0 (fun=0x3f94ee70) at ../py/runtime.c:598
#11 0x400e148d in parse_compile_execute (source=0x3f94ed5e, input_kind=MP_PARSE_FILE_INPUT, exec_flags=32) at ../lib/utils/pyexec.c:103
#12 0x400e1710 in pyexec_file (filename=0x3f94ed5e <error: Cannot access memory at address 0x3f94ed5e>) at ../lib/utils/pyexec.c:560
#13 0x400e01ca in TASK_Micropython (pvParameters=<optimized out>) at mptask.c:320
```

closed 2021-08-18 by russhughes