russhughes/st7789_mpy #17: wasp-os Twatch-2020 (sactre repo) with this repo : no enough ram

picpic020960, 2021-02-22:
Bonjour ,
test as say in title. 
after little adapts ( SPI order parameter , '.string' to '.text' ) the 'booting' message is watch screen displayed.
but after :
Traceback (most recent call last):
  File "wasp.py", line 383, in run
  File "uasyncio/core.py", line 1, in run
  File "uasyncio/core.py", line 1, in run_until_complete
  File "uasyncio/core.py", line 1, in run_until_complete
  File "wasp.py", line 393, in main_loop
  File "wasp.py", line 114, in init_apps
MemoryError: memory allocation failed, allocating 640 bytes
and reboot.

I see that this patch use some memory for RGB and compress.
Is it possible one version with st7789 driver only , compatible with draw565.py
of twatch-2020 ?
PS : tested so micropython-lvgl but too hard to adapt to SACTRE wasp-os repo.  

thanks for help and
Regards

russhughes, 2021-02-22:
I will look into this.

picpic020960, 2021-03-13:
Bonjour Russ
OK for the ram after clean 
now try to compile all micropython with st7789 but KO.
success for micropython 1.14 
i see the prebuild firmware is 5 days old.
how do you do ?

test with make or idf.py , some messages : 

make : 

xxxxx@LIN0141157:~/micropython/ports/esp32$ make -DMODULE_ST7789_ENABLED=1 USER_C_MODULES=../../../st7789_mpy/ all
make : option invalide -- 'D'
make : option invalide -- 'M'

idf.py : 

-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    MODULE_ST7789_ENABLED

...
-- Build files have been written to: /home/xxxxx/micropython/ports/esp32/build
ninja: error: unknown target '../st7789_mpy'
command "USER_C_MODULES=../../../st7789_mpy/" is not known to idf.py and is not a Ninja target

thank's for helps
Regards

russhughes, 2021-03-13:
The module does not compile with a CMake build see https://github.com/russhughes/st7789_mpy/issues/24 for more information.

closed 2021-03-13 by russhughes