russhughes/st7789_mpy #23: on/off methods

jej, 2021-03-08:
Hi Russ,
I see in the devbis version the on() and off() methods. Did you remove it or was it pushed after you forked? Is there any interest to have these methods?
Thx

```
STATIC mp_obj_t st7789_ST7789_on(mp_obj_t self_in) {
    st7789_ST7789_obj_t *self = MP_OBJ_TO_PTR(self_in);
    DISP_HIGH();
    mp_hal_delay_ms(10);

    return mp_const_none;
}

STATIC mp_obj_t st7789_ST7789_off(mp_obj_t self_in) {
    st7789_ST7789_obj_t *self = MP_OBJ_TO_PTR(self_in);
    DISP_LOW();
    mp_hal_delay_ms(10);

    return mp_const_none;
}
```

russhughes, 2021-03-08:
Looks they we added after I forked, I will add the methods.

russhughes, 2021-03-09:
The on and off methods to control the backlight pin are now available.

closed 2021-03-09 by russhughes