russhughes/st7789_mpy #24: Problem with building external C modules

ypj0202, 2021-03-08:
I encounter a problem when adding external C modules, I check the official documentation and use the command `make USER_C_MODULES=../../../st7789_mpy/ all CFLAGS_EXTRA=-DMODULE_ST7789_ENABLED=1` . The firmware can be compile with no errors, but st7789 is not being complied.

Version I used
esp-idf = 4.1.1
gcc8.4.0
Micropython 1.14

russhughes, 2021-03-08:
This is probably related to the recent move to a CMake build system.  See https://forum.micropython.org/viewtopic.php?f=18&t=9820 and/or https://github.com/russhughes/st7789_mpy/issues/8 for more information.  

Once a method of adding external C modules is worked out I will support it.

ypj0202, 2021-03-08:
Thanks for the information, another question that I would like to ask: For ESP32 port, in which file the port name is defined? Also the version number?

russhughes, 2021-03-08:
I'm not sure if this is correct for the CMake builds. For the previous make build system:

The port's name is defined in the `mpconfigboard.h`  in the ports `boards` subdirectory for example: `micropython/ports/esp32/boards/GENERIC/mpconfigboard.h`.  

The version number is created at compile time by the `micropython/py/makeversionhdr.py` script and saved in the build directory file `genhdr/mpversion.h` and referenced in the `moduos.c` file in the `ports/esp32` directory.

ypj0202, 2021-03-08:
Last question, if I got a custom board with ESP32 and I want to build  firmware for it, I just make a new folder with `mpconfignpard.cmake` ,`mpconfigboard.h` and custom modules folder, is this how you make your custom board? Correct me if I am wrong. Are there any documentation I can look for building firmware for custom ESP32 board(Create a new board in `/ports/esp32/boards`)? Thanks in advanced!

russhughes, 2021-03-08:
Yes. You can override things using your own custom boards. 

For example my M5CORE2 board has:
manifest.py  modules/  mpconfigboard.h  mpconfigboard.mk  sdkconfig.board

manifest.py:
include("$(PORT_DIR)/boards/manifest.py")
freeze("modules")

mpconfigboard.h:
include("$(PORT_DIR)/boards/manifest.py")
freeze("modules")

mpconfigboard.mk:
SDKCONFIG += boards/sdkconfig.base
SDKCONFIG += boards/sdkconfig.spiram
FROZEN_MANIFEST ?= $(BOARD_DIR)/manifest.py

sdkconfig.board:
CONFIG_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_ESP32_REV_MIN_1=y

moldules directory contains python files to be frozen for this board.  I don't know if there is a better or more correct method, but this one works.

closed 2021-03-08 by russhughes