russhughes/st7789_mpy #25: multiple display function

HorwinHe, 2021-03-16:
hi! I know multiple display can use TFT-eSPI, but I don't know C language, how can I use more than two display with this driver. Can you tell me how to do it.  thanks bro.

russhughes, 2021-03-21:
I have not tested this myself, but you should be able assign each display's CS to a different pin.  

```spi = SPI(2, baudrate=30000000, sck=Pin(18), mosi=Pin(19))

# initialize 1st display
tft1 = st7789.ST7789(
    spi,
    135,
    240,
    reset=Pin(23, Pin.OUT),
    cs=Pin(5, Pin.OUT),
    dc=Pin(16, Pin.OUT),
    backlight=Pin(4, Pin.OUT),
    rotation=0,
    buffer_size=64*64*2)

# enable 1st display and clear screen
tft1.init()
tft1.fill(st7789.BLACK)

# initialize 2nd display
tft2 = st7789.ST7789(
    spi,
    135,
    240,
    reset=Pin(23, Pin.OUT),
    cs=Pin(6, Pin.OUT),
    dc=Pin(16, Pin.OUT),
    backlight=Pin(4, Pin.OUT),
    rotation=0,
    buffer_size=64*64*2)

# enable 2nd display and clear screen
tft2.init()
tft2.fill(st7789.BLACK)
```

HorwinHe, 2021-03-22:
> 我自己没有对此进行测试，但是您应该可以将每个显示器的CS分配给不同的引脚。
> 
> ```
> 
> # initialize 1st display
> tft1 = st7789.ST7789(
>     spi,
>     135,
>     240,
>     reset=Pin(23, Pin.OUT),
>     cs=Pin(5, Pin.OUT),
>     dc=Pin(16, Pin.OUT),
>     backlight=Pin(4, Pin.OUT),
>     rotation=0,
>     buffer_size=64*64*2)
> 
> # enable 1st display and clear screen
> tft1.init()
> tft1.fill(st7789.BLACK)
> 
> # initialize 2nd display
> tft2 = st7789.ST7789(
>     spi,
>     135,
>     240,
>     reset=Pin(23, Pin.OUT),
>     cs=Pin(6, Pin.OUT),
>     dc=Pin(16, Pin.OUT),
>     backlight=Pin(4, Pin.OUT),
>     rotation=0,
>     buffer_size=64*64*2)
> 
> # enable 2nd display and clear screen
> tft2.init()
> tft2.fill(st7789.BLACK)
> ```

First of all, thank you for your reply. I try to configure it like you do, but it's strange that after the initialization of both displays, I can't update the display through TFT1. Fill() or TFT1. Text(). I have to add TFT1. Init(). If I do this, the content displayed on the other display will disappear. I try to pull up the CS pin of the display that I don't operate from the code, But that doesn't seem to work. I feel very confused.

russhughes, 2021-03-22:
I have two displays on order so I can work on the issue later this week.

HorwinHe, 2021-03-31:
hi, bro, Sorry to disturb you for a long time, have you tested it? I'm looking forward to your test results. How can I contact you.

russhughes, 2021-04-01:
I have verified the driver works with two displays.  The two displays can share the same Vcc, Gnd, Data in,  Clock, DC (Data/Cmd) , and BL (Backlight) pins.  The CS (Chip Select)  of each display must be connected to different pins so the driver is able to select to each display independently.   

If the displays CS pins are connected to the same pin the driver will write the same thing to both displays.  

If the Reset pins are connected to the same pin, the first display will stop working when the second display is initialized because it will see the reset signal when the second display is initialized.  You can work around this by only specifying the reset pin for the first display and not specifying a reset pin on the other displays. This will reset all displays when you initialize the first display.  If you need to be able to reset each display independently, they must have their own reset pin.

You should be able to drive as many displays as you like as long as they have their own CS pins.

closed 2021-04-04 by russhughes