russhughes/st7789_mpy #26: map_bitarray_to_rgb565

vslavav, 2021-03-16:
Hi
I have problem with function: map_bitarray_to_rgb565

buffer = bytearray(512)
bm_buffer = bytearray(32)
    
    try:
        memoryview(buffer)
        memoryview(bm_buffer)
        print('support')
    except TypeError:
        print('not support')

always print 'support'

but when i call 
st7789.map_bitarray_to_rgb565(bm_buffer, buffer, 16, st7789.YELLOW,st7789.BLACK)

i get
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "map_bitarray_to_rgb565_ex.py", line 154, in <module>
  File "map_bitarray_to_rgb565_ex.py", line 150, in test
TypeError: object with buffer protocol required

please help

vslavav, 2021-03-16:
fixed

closed 2021-03-16 by vslavav

psyc0, 2021-03-21:
> fixed

how? Im trying the same for bitmapped fonts so was just wondering if it was something similar you where doing

vslavav, 2021-03-22:
> > fixed
> 
> how? Im trying the same for bitmapped fonts so was just wondering if it was something similar you where doing

I used this function from module level. It was a mistake. You need to use it from Class level.
Below is example of how to use Peter Hinch fonts (micropython-font-to-py) :

def printString_fb(string,display,col_start, row_start, fg=st7789.WHITE, bg=st7789.BLACK):   # using framebuffer
   
    
    for char in string:
        
        glyph, char_height, char_width = font.get_ch(char) 
        buf = bytearray_at(addressof(glyph), len(glyph))
        
        buffer = bytearray(char_width*char_height*2)    
        display.map_bitarray_to_rgb565(buf, buffer, char_width, st7789.YELLOW,st7789.BLACK) 
                

              
        display.blit_buffer(buffer, col_start, row_start, char_width, char_height)
        col_start += char_width

psyc0, 2021-03-23:
Thanks, it was actually exactly what I was after as well!.

bjsiyong, 2024-01-31:
> Thanks, it was actually exactly what I was after as well!.

does it work?