russhughes/st7789_mpy #27: Custom fonts example with font-to-py

psyc0, 2021-03-21:
Hi,

From devbis original fork there is a reference to font-to-py, I had some issues getting it to work but probably due to incorrect assumptions on my end. I created and discussed this with Peter over in his repo. [font-to-py issue 34](https://github.com/peterhinch/micropython-font-to-py/issues/34)

If possible, an example with these types of custom fonts would be nice, and if you have any tips/hints to give me as to what I'm doing wrong, that would be greatly appreciated.

russhughes, 2021-03-21:
I'm working on a font to python utility for the st7789 and ili9342 drivers today.

psyc0, 2021-03-21:
Great news!, I will surely try it as soon as its available.

closed 2021-03-21 by psyc0

russhughes, 2021-03-24:
Utility and examples for monospaced True Type fonts are up.