russhughes/st7789_mpy #28: TFT 240x240 : OK with Arduino lib with 4 wires , KO with st7789_mpy

picpic020960, 2021-03-25:
Bonjour Russ ,
have a TFT 240x240 
connect to ESP32 with 4 wires :+1:  
 SCL : 18
 SDA : 23
 RES : 4
 DC   : 2
arduino lib Bodmer/TFT_eSPI-2.3.59.zip , example program : work fine.

then load micropython with st7789 c driver (.bin T-DISPLAY feb) , transpose the pins in the hello.py , run ...
and nothing on the screen.

How can i debug ?
Thanks for help

robhamerling, 2021-03-25:
I have a similar display (but 8 pins),  it works fine with ESP32 and Micropython (latest firmware).
 ```
spi = SPI(2, baudrate=30000000, polarity=1, phase=1, sck=Pin(18), mosi=Pin(23))
tft = st7789.ST7789(spi, 240, 240,
                     reset=Pin(22, Pin.OUT),
                     cs=Pin(5, Pin.OUT),
                     dc=Pin(21, Pin.OUT),
                     backlight=Pin(4, Pin.OUT),
                     rotation=0)            
```

Remark: I find it somewhat confusing that the board uses PIN names SCL and SDA. These terms are normally used with I2C, while the ST7789 uses SPI protocol and SCK and SDI/MOSI would be appropriate.

picpic020960, 2021-03-25:
Hello robhamerling
works fine with jour parameters
thanks a lot

closed 2021-03-25 by russhughes