russhughes/st7789_mpy #29: jpg (96x60) ok but position x,y ko

picpic020960, 2021-03-27:
Bonjour Russ ,
try png to jpg online converters :OK (but not all)
then test with jpg.py 
the icon is displayed at 0,0 position
but  when change to x,y the icon is always at 0,0
how to debug ?
thanks for help
Regards

PS : (so can't test many jpg on the display ...)

russhughes, 2021-03-28:
I added conversion instructions and another jpg example for the T-Display.  What device are you using?

picpic020960, 2021-03-28:
Thanks Russ
Will test tomorrow

russhughes, 2021-04-04:
The buffer should be large enough to hold the bitmap (height * width * 2) when using the FAST method.  You can leave the buffer_size out and the driver will allocate and free memory as needed. Using the buffer_size trades memory for speed.

I’ll be updating the driver to work with the new Cmake build system this week.

russhughes, 2021-04-10:
The driver can now be compiled using the new CMake build system.

closed 2021-04-10 by russhughes