russhughes/st7789_mpy #30: "Clone this module alongside the MPY sources:  $ git clone ..."  ???

picpic020960, 2021-04-14:
Bonjour Russ ,
I try to compile st7789_mpy module with cmake micropython version.
cmake micropython is OK
but for clone st7789_mpy repo I don't know where : 
what folder ?
new folder ?
frozen_mpy ?
Thank's for help
Regards

russhughes, 2021-04-15:
cd your_directory_you_want_to_use
git clone https://github.com/micropython/micropython.git
git clone https://github.com/russhughes/st7789_mpy.git

Making both repos subdirectories of the same directory for example:
/home/russ/micropython
/home/russ/st7789_mpy

The 'user modules directory'  can be used when you have several C external modules you want to include at the same time. You  could with the old make build system (not sure if this works with the cmake build system) create a directory and place several modules in it . You could then specify the directory as the user modules directory and the build system would include each of the modules. 

for example:
/home/russ/my_micropython-modules/st7789_mpy
/home/russ/my_micropython-modules/ili9342c_mpy

picpic020960, 2021-04-15:
Thanks Russ
I go on

closed 2021-04-15 by russhughes