russhughes/st7789_mpy #32: README additional info

robhamerling, 2021-04-27:
1. The README file mentions as optional arguments among others 'dc'. But when I do not specify dc I get a message: ValueError: must specify dc pin

2. A suggestion might be added about what to do with cs, blk, res when present at the display but  not wired (and not specified in the program)? Pulled high/low,..?

russhughes, 2021-04-27:
I agree those details should be cleaned up and made clearer. 

The `dc`  pin is definitely not optional, it is the pin that identifies display data/command selection .  The parameter may have originally had a default value making the parameter optional but the pin was always required. 

`cs` can be tied Low if the display is the only device on the SPI port.  The display will always be the selected device. 

`res` can be tied high. You will not be able to perform a hard reset, only software resets.

`blk` can be left floating or disconnected as there is usually a pull up resistor on the display module. The backlight will aways be powered on and cannot be turned off.

Display modules vary between vendors and models so these are general guide lines.

robhamerling, 2021-04-28:
Looks OK to me. 
FYI: On my board I have now 'res' and 'blk' of the ST7789 permanently pulled high and 'dc' connected to the (free) MISO line of the SPI port. Which results (with active 'cs') a 6-wire connection in stead of 8-wires.

robhamerling, 2021-05-02:
Another advice / warning might be added:
I hadn't touched my ESP32 TTGO T-Display V1.1 for a while, but it worked well with your pre-compiled firmware 1.14. Today I loaded my self-compiled version1.15-54. Works well with an ESP32 with separate ST7789 display using hardware SPI(2), but not with the TTGO board with ST7789: the display kept black.   After some analysis and experiments I discovered that on this board MOSI for the display is wired to Pin(19) which is not standard for hardware SPI(2), for which it is Pin(23). 
It seems the hardware SPI interface has become more strict/rigid w.r.t. Pin assignments.
After I switched to SoftSPI the display became alive again!
Note: SoftSPI requires specification of MISO, so I specified an unused pin.

russhughes, 2021-05-05:
Thanks for reminding me of the SPI changes. I'm going to work on bringing the examples and documentation up to date this weekend.

robhamerling, 2021-06-13:
I had successfully built Micropython firmware 1.15 (211) for the Teensy40  (Linux Mint 20.1)
Now want to add the ST7789 driver.
The  README tells me to add among others the make-option:
 `-DMODULE_ST7789_ENABLED=1`
 but make responds with:
 `  invalid option -- 'D'`

Without this option makes seems to compile the st77789 code, but there is a link problem:
`LINK build-TEENSY40/firmware.elf`
`arm-none-eabi-ld: build-TEENSY40/st7789/st7789.o: in function 'st7789_ST7789_draw':`
`st7789.c:(.text.st7789_ST7789_draw+0x11c): undefined reference to 'lround'`
`arm-none-eabi-ld: st7789.c:(.text.st7789_ST7789_draw+0x138): undefined reference to 'lround'`
`arm-none-eabi-ld: st7789.c:(.text.st7789_ST7789_draw+0x1a0): undefined reference to 'lround'`
`arm-none-eabi-ld: st7789.c:(.text.st7789_ST7789_draw+0x1ba): undefined reference to 'lround'`

This doesn't appear when I build firmware with st7889 driver for ESP32 and STM32 boards. 
What am I missing?

russhughes, 2021-06-13:
The TEENSY library must not have the lround math function or it may have to be linked from a math library.  A quick fix is to remove or comment out lines 424, 426, 427, and 428 leaving line 425 in the st7789.c file.

Before: 

https://github.com/russhughes/st7789_mpy/blob/fc8d69dc5f66bdac09a444059308031658e2a497/st7789/st7789.c#L424-L428

After:
```
//#ifdef MICROPY_PY_STM
#define LROUND(x) ((long) roundf(x))
//#else
//#define LROUND(x) (lround(x))
//#endif
```

robhamerling, 2021-06-13:
I followed your instructions, but I'm afraid it results in error messages like this one:

[image]

Note: I took a screenshot to preserve formatting.

robhamerling, 2021-06-14:
I got rid of the error msgs by using 'round' in stead of 'roundf', thus:
`#define LROUND(x) ((long) round(x))`
which resulted in a firmware version for the Teensy40 with a working st7789 module!

russhughes, 2021-06-14:
Excellent, I will make a note to fix that in the next release.

closed 2021-08-14 by russhughes