russhughes/st7789_mpy #34: Another advice / warning might be added:

AngainorDev, 2021-06-20:
Another advice / warning might be added:
I hadn't touched my ESP32 TTGO T-Display V1.1 for a while, but it worked well with your pre-compiled firmware 1.14. Today I loaded my self-compiled version1.15-54. Works well with an ESP32 with separate ST7789 display using hardware SPI(2), but not with the TTGO board with ST7789: the display kept black.   After some analysis and experiments I discovered that on this board MOSI for the display is wired to Pin(19) which is not standard for hardware SPI(2), for which it is Pin(23). 
It seems the hardware SPI interface has become more strict/rigid w.r.t. Pin assignments.
After I switched to SoftSPI the display became alive again!
Note: SoftSPI requires specification of MISO, so I specified an unused pin.

_Originally posted by @robhamerling in https://github.com/russhughes/st7789_mpy/issues/32#issuecomment-830804791_

AngainorDev, 2021-06-20:
Had the same issue, MPY 1.16 and TTGO T-Display v1.1, esp-idf V4.2

No visible error but black screen.  
Switching to SoftSPI and using Pin(14) as MISO revived the display, but it's significantly slower to refresh.

Kept the same 30000000 as baudrate, pushed the ESP32 to 240Mhz, still slower than Hardware SPI was on older MicroPythons.

Any hope to get hardware SPI working again with the TT-Display pins?

AngainorDev, 2021-06-20:
Updating after a "haha" moment and re-read of the mpy docs, keeping the issue open for reference and readme/examples update.

mpy doc states other pins than default can be used, just will go through an extra multiplexing layer.
Old code did override mosi but not miso. On newest mpy, you need to explicitely specify miso as well.

I got it running with hardware SPI and these params:    
`SPI(2, baudrate=30000000, polarity=1, phase=1, sck=Pin(18), mosi=Pin(19), miso=Pin(14))`

14 is an arbitrary unused pin I picked.

robhamerling, 2021-06-20:
Back to SPI(2) to which I added miso=Pin(37) and the display works fine again! ( MP 1.16-2 )
Thanks.

closed 2021-08-14 by russhughes