russhughes/st7789_mpy #36: Error using bitmap method

etolocka, 2021-07-25:
Hello
I have used your firmware for a project that uses the TTGO Display and I have been able to get all the methods to work successfully but I have a problem using bitmap.
In the following code I use the already converted t1.py and t2.py modules found in the toasters / folder but I always get the error "TypeError: can't convert str to int" What am I doing wrong? Thank you very much for your help and for the excellent work you have done with the driver.

from machine import Pin, SPI
import st7789

import t1, t2

tostadas = [t1, t2]

spi = SPI(1, baudrate=30000000, polarity=1, phase=1, sck=Pin(18), mosi=Pin(19))
display = st7789.ST7789(
    spi, 
    135,
    240,
    reset=Pin(23, Pin.OUT),
    cs=Pin(5, Pin.OUT),
    dc=Pin(16, Pin.OUT),
    backlight=Pin(4, Pin.OUT),
    rotation=0,
    buffer_size=64*64*2)

display.init ()

display.bitmap (tostadas[0], 0, 0)

etolocka, 2021-07-26:
Hi there!
I have continued testing the example program and modifying it by removing functionalities so that it looks like the one I did originally and it strikes me that I can call the bitmap function in the same way as in the program I sent before and the error does not appear. What confuses me is that the error "TypeError: can't convert str to int" starts to appear when I remove the definition of sprites with the three objects inside, which I don't understand because those objects are not used in the call to bitmap, which invokes the module directly. Even if I remove one toast object definition and leave the other two, the error starts to appear.

The following code runs ok with no errors:

import time
import random
from machine import Pin, SPI
import st7789
import t1,t2,t3,t4,t5

TOASTERS = [t1, t2, t3, t4]
TOAST = [t5]

class toast():
    '''
    toast class to keep track of a sprites locaton and step
    '''
    def __init__(self, sprites, x, y):
        pass
    

def main():
    '''
    Draw and move sprite
    '''
    spi = SPI(1, baudrate=30000000, sck=Pin(18), mosi=Pin(19))

    # initialize display
    tft = st7789.ST7789(
        spi,
        135,
        240,
        reset=Pin(23, Pin.OUT),
        cs=Pin(5, Pin.OUT),
        dc=Pin(16, Pin.OUT),
        backlight=Pin(4, Pin.OUT),
        rotation=0,
        buffer_size=64*64*2)

    # enable display and clear screen
    tft.init()
    tft.fill(st7789.BLACK)

    # create toast spites in random positions
    sprites = [
        toast(TOASTERS, 135-64, 0),
        toast(TOASTERS, 135-64, 0),
        toast(TOASTERS, 135-64*4, 160)
    ]

    # move and draw sprites
    tft.bitmap(t1, 0, 0)
    

main()

russhughes, 2021-07-26:
I will look at this later tonight.

russhughes, 2021-07-27:
This is definitely a bug in the driver when getting the parameters. You can work around the issue for now by adding the optional index parameter with a value of 0.  For example:
display.bitmap(tostadas[0], 0, 0, 0)

Russ

etolocka, 2021-07-27:
Thanks! With the index parameter at 0 it works correctly

closed 2021-07-29 by russhughes