russhughes/st7789_mpy #37: floating point conversion errors

robhamerling, 2021-08-15:
When building version 0.2.1 for the TEENSY 4.0 several error messages like the following appear:

`/home/rob/st7789_mpy/st7789/st7789.c:493:21: error: conversion from 'double' to 'int16_t' {aka 'short int'} may change value [-Werror=float-conversion]
`

As a result no firmware with support for ST7789.

robhamerling, 2021-08-15:
After a provisional fix for the floating point conversion another error messages appeared:

`arm-none-eabi-ld: build-TEENSY40/st7789/st7789.o: in function `st7789_ST7789_fill_polygon':
st7789.c:(.text.st7789_ST7789_fill_polygon+0x1d2): undefined reference to `qsort'
`

russhughes, 2021-08-15:
I'm able to get a clean TEENSY 4.0 build now but don't have a board to test the firmware.

robhamerling, 2021-08-16:
Until I installed the st7789 version with support for polygons I had no problems with building the TEENSY 4.0 firmware with ST7789 module. 
I provisionally 'solved' the issue with floating point conversion by removing "-Wfloat-conversion" from the Makefile. Maybe the issue with "missing qsort" is a shortcoming with my install of the arm compiler/linker. If you can build the firmware there must be a difference between your and my system, but what? I'm prepared to test your firmware.
BTW Don't know if it is relevant but I'm working with Linux Mint 20.2.

russhughes, 2021-08-16:
Sorry I meant that I get a clean build with the latest changes I’ve made to the driver. It looks like the libc library did not include the qsort routine so I’ve reverted back to the trusty bubble sort used in the original polygon code and fixed the floating point conversations.  Pull the latest version and it should build.

robhamerling, 2021-08-16:
Both issues solved! Great, thanks!

closed 2021-08-16 by russhughes