russhughes/st7789_mpy #40: No BG color

jdtsmith, 2021-08-24:
One of the nice features of the loboris port's `display` module is a font option of:

```
transparent | only draw font's foreground pixels
```

This would be wonderful to have in the `write`/`text`/`draw` text primitives, perhaps by recognizing `bg=None`.  This allows you to write text atop images and other markings.

russhughes, 2021-08-24:
The draw method uses lines that have no background, so it should have a transparent background.  

The write and text methods write each character to a single character sized buffer then blit that buffer to the display.  This is much faster then plotting each pixel of a character individually.  To save RAM the driver does not use a frame buffer. There is no way to determine what colors are on the display to make a "transparent" background. The only way to implement transparency in this case would be to plot each pixel individually when given a bg=none. I will look into this.

jdtsmith, 2021-08-24:
That makes sense, thanks.  As you saw the draw method is what is giving me problems in #39.  I checked out how the Loboris port does it and it effectively just sets each FG pixel one at a time when font transparency is desired, skipping BG pixels, and sets and blits an entire char buffer otherwise, as you mention here.  Slow, but it's been effective for my needs.  I presume vector fonts do something like this as well. 

I coded up a simple thought exercise along these lines for `text`; see [this commit](https://github.com/jdtsmith/st7789_mpy/commit/9acb33282363bba1cf4c494726bcd5c42e479704) on my fork.  I haven't tested it at all.  The build process seems pretty involved, and I'm not sure what additional build steps you are taking for the various firmwares, but I'll see what I come up with (unless you get there first!). 

Thanks again!

jdtsmith, 2021-08-24:
Well, I got it to compile and loaded the new generic firmware, but it uses about 40K more memory than your firmware (not sure why), which makes it impossible for me to use (can't load an image!).  But, it works!  E.g. `tft.text(vga,"It Works!!!",0,5+vga.HEIGHT,0x0ff0,None)`

[image]

That said, it is really very slow, much slower than Loboris' similar functionality.  Probably he's optimized individual pixel updates somehow.  I don't suppose you can afford memory-wise to keep a blit buffer hanging around with the latest image data, so you can just update the relevant pixels, subset it, and send it over SPI en masse?

jdtsmith, 2021-08-24:
[Bingo](https://github.com/loboris/ESP32_TFT_library/blob/aa21772f54a71887ec08b2e8bbaef9e304009891/components/tft/tftspi.c#L463).  So perhaps we could pull a piece of the screen over SPI (for one character), update all the FG pixels, and then send it back.

russhughes, 2021-08-24:
It would take 64800 bytes to maintain a 16 bit full frame buffer (240x135x2).  An alternative would be to maintain a  smaller color depth buffer and limit the number of colors that could be used.  For example a 4 bit color depth would allow for 16 colors and require 240x135x0.5 or 16200 bytes.  An 8 bit buffer that would allow for 256 colors requires 32400 bytes for the frame buffer and 512 bytes for the color palette table. Both would be slower then the existing methods but would be much faster then the per pixel method.

The SPI interface does not allow you to read back the screen buffer, this can only be done with an 8 bit parallel interface. This of course would have solved the issue if it were available.  

I'm considering a rewrite of the driver to combine support for different display controllers instead of maintaining separate drivers. I might consider adding frame buffer support at that time.

jdtsmith, 2021-08-24:
Huh, I wonder why loboris port seems to think it's possible to read back from the display RAM?  In any case it looks like they are banging bits.  Not sure why it is faster on the TTGO given that.  It's interesting that the ST7789 data sheet says:

> Display data can be stored in the on-chip display data RAM of 240x320x18 bits. It can perform display data RAM read/write operation with no external operation clock to minimize power consumption.

I guess accessing that via the `RAMRD` command is only possible via the parallel interface you mention?

For frozen in font glyphs, an alternative would be to efficiently "pack" a given glyph into a hopefully small set of rectangular regions or lines, and send those in bulk.  This could happen offline, i.e. in the compiled .mpy frozen font file (which would no longer be a bitmap, but a rectangle list).  That seems similar to and probably a bit more efficient I'd guess to how Hershey drawing works (with lots of vertical/horizontal lines).  The generalized delta method (left) looks reasonably straightforward:

[image]

[Here's an interesting](https://github.com/jnfran92/adaptive-boxes) Python module that implements another rectangle decomposition algorithm.  You could of course hard-code and freeze in any number of bitmaps this way, not just font glyphs.  Fun stuff.

russhughes, 2021-08-24:
You can expand the application partition in the flash by editing the partitions.csv file in ports/esp32. The default partitions.csv file uses an application partition size of 0x180000 that is large enough for the stock MicroPython image.  You can increase the application partition to 0x1e0000 by modifying the partitions.csv file and recompiling. When you change partition sizes you should erase the flash before flashing the modified firmware.

default  partitions.csv file
```
# Notes: the offset of the partition table itself is set in
# $IDF_PATH/components/partition_table/Kconfig.projbuild.
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x180000,
vfs,      data, fat,     0x200000, 0x200000,
```

modified  partitions.csv file
```
# Notes: the offset of the partition table itself is set in
# $IDF_PATH/components/partition_table/Kconfig.projbuild.
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x1e0000,
vfs,      data, fat,     0x200000, 0x200000,
```

if you have one of the 16MB flash T-DISPLAYs you can expand the vfs data partition as well.

```
# Notes: the offset of the partition table itself is set in
# $ESPIDF/components/partition_table/Kconfig.projbuild and the
# offset of the factory/ota_0 partition is set in makeimg.py
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x1e0000,
vfs,      data, fat,     0x200000, 0xe00000,

```

closed 2021-11-25 by jdtsmith