russhughes/st7789_mpy #49: build error

heixiaoma, 2021-11-12:
```
bootloader  @0x001000    23216  (    5456 remaining)
partitions  @0x008000     3072  (    1024 remaining)
application @0x010000  2386544  (  -27248 remaining)
ERROR: application overflows allocated space of 2359296 bytes by 27248 bytes
make: *** [Makefile:35：all] 错误 1

```

russhughes, 2021-11-12:
Which device are you using?

heixiaoma, 2021-11-13:
esp32 8M psram

heixiaoma, 2021-11-13:
make add BOARD=GENERIC_SPIRAM  Param

russhughes, 2021-11-13:
You need to increase the size of the application partition to accommodate the the additional driver code.  You can edit the partitions.csv file in the micropython/port/esp32 directory and recompile to reserve more flash space for MicroPython.  Whenever you change the partitions.csv table and recompile, you must erase before flashing the new firmware.

For example:
```# Notes: the offset of the partition table itself is set in
# $IDF_PATH/components/partition_table/Kconfig.projbuild.
# Name, Type, SubType, Offset, Size, Flags
nvs, data, nvs, 0x9000, 0x6000,
phy_init, data, phy, 0xf000, 0x1000,
factory, app, factory, 0x10000, 0x247000,
vfs, data, fat, 0x257000, 0x1A7000,
``` 

See my post on [stack overflow](https://stackoverflow.com/a/69277852/2259205) for more details. If your device has more than 4MB of flash you can increase the size of the vfs partition as well.

closed 2021-11-25 by russhughes