russhughes/st7789_mpy #50: lv_micropython build failed

Meekdai, 2021-11-26:
I used the latest lv_micropython to build the firmware containing st7789_mpy, but an error occurred.
Here is my compilation log: [https://github.com/Meekdai/lv_micropython-actions/runs/4331465046?check_suite_focus=true](url)

This may be the cause of the error:[https://github.com/lvgl/lvgl/commit/eaf25348a70d995ae0b19182ffbe4b966f5a6b20](url)

russhughes, 2021-11-26:
Both LVGL and the st7789 driver use the tjpgd565 library and each are including their own copy causing duplicate functions with the same name.  You might be able to resolve the issue by removing tjpgd565.c from the st7789 driver directory and any references to it in the cmake files.  There are differences in the tjpgd565.h file, my header file has two additional variables to support the JPG slow draw method. 

I have not used this driver with LVGL.  I use a [modified LVGL ili9XXX driver](https://github.com/russhughes/lv_st7789)  with LVGL . Did this successfully build before they made the update?  If not it, it might be better to use the modified ili9XXX driver.

Meekdai, 2021-11-27:
Now I have solved the issue by following your method.
It successfully builded. [log](https://github.com/Meekdai/lv_micropython-actions/actions/runs/1510358315)

closed 2021-11-27 by Meekdai