russhughes/st7789_mpy #51: building micropython-firmware for esp32 failed

khvolk, 2021-12-05:
I just tried to build the micropython-firmware for esp32 with the four modules:

  NotoSans_32.py
  scripts.py
  vga1_16x16.py
  vga1_16x32.py

added: 

make USER_C_MODULES=../../../../st7789_mpy/st7789/micropython.cmake FROZEN_MANIFEST="" FROZEN_MPY_DIR=$UPYDIR/modules 

failed with:
  ...
  FAILED: genhdr/qstr.i.last
  ...
  /home/kh/micropython/ports/esp32/modmachine.c:40:10: fatal error: esp32/clk.h: No such file or directory
 #include "esp32/clk.h"

[firmware-compile.txt](https://github.com/russhughes/st7789_mpy/files/7655756/firmware-compile.txt)
 
The machine was Ubuntu 20.04.3. Any idea?

russhughes, 2021-12-05:
I don't know what would cause that error. I'm not able to read the firmware-compile.txt file it downloads only 2 bytes.

toymakerlabs, 2021-12-06:
Interesting, I get this error too when trying to build for the esp32. I'm wondering if this might be upstream in Micropython, or an esp-idf + micropython versioning thing. I came across this from looking around: [ESP-IDF 5.0 Migration guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/system.html. The guide mentions that `esp32/clk.h` is deprecated. I tried switching `micropython/ports/esp32/modmachine.c` line 40 from:

`#include "esp32/clk.h"`
to
`#include "esp_private/esp_clk.h"`

Which got me past that error. I have other errors I'm working through, though I believe they are micropython errors and not related to the st7789 driver lib here. (Though, this is coming from a total novice) Definitely appreciate the driver and the setup guide! 

Cheers

russhughes, 2021-12-06:
That make sense.

khvolk, 2021-12-06:
This new #include "esp_private/esp_clk.h" in micropython/ports/esp32/modmachine.c line 40 worked with this missing include. But now I ran into another problem: linking together to micropython.elf fails with lots of 

  dangerous relocation: call8: call target out of range, e.g.:

CMakeFiles/micropython.elf.dir/home/kh/st7789_mpy/st7789/st7789.c.obj: in function `st7789_ST7789_draw':
st7789.c:(.text+0xf23): dangerous relocation: call8: call target out of range: mp_obj_float_get
st7789.c:(.text+0x1073): dangerous relocation: call8: call target out of range: __extendsfdf2
st7789.c:(.text+0x1085): dangerous relocation: call8: call target out of range: __adddf3

I try again to insert my log-file

[second.txt](https://github.com/russhughes/st7789_mpy/files/7660346/second.txt)

using drag and drop (from my manjaro/plasma desktop)

khvolk, 2021-12-06:
This happens even, if I remove my added files NotoSans_32.py, scripts.py, vga1_16x16.py and vga1_16x32.py from "modules/" and doing make clean.

fkrenn12, 2021-12-23:
I also got the described error above. I tried it  several times on a WSL Ubuntu 20.04 LTS. It seems, that the micropython source is not compatible with the latest esp-idf on github. 
A step back to the idf4.2 release solved it.

git fetch
git checkout release/v4.2
git pull
git submodule update --init --recursive

After a "make clean" a new make it was successfully build.

closed 2021-12-23 by russhughes

russhughes, 2021-12-23:
Thanks for the information, I've updated the README.md to use release v4.2

commit a4287945 (2021-12-23): Update README.md to fix #51