russhughes/st7789_mpy #52: build for port STM32L496G-DISC

patfrench, 2021-12-31:
Hi,

When i build in :
```cd micropython/ports/stm32```
``make USER_C_MODULES=../../../st7789_mpy/st7789/micropython.cmake``

I have a pyboard by default.
how to do the same for the board STM32L496G-DISC ?

Thank you

russhughes, 2022-01-01:
The math library uses doubles for sin and cos. I'll see what I can do to fix this, until then, you can get around the error by changing sin and cos to sinf and cosf if they are supported. Failing that you should be able to cast the results of sin and cos to floats. 

```
mp_float_t cosAngle = cosf(angle);
mp_float_t sinAngle = sinf(angle);
```

or possibly:

```
mp_float_t cosAngle = (mp_float_t) cos(angle);
mp_float_t sinAngle = (mp_float_t) sin(angle);
```

Russ

patfrench, 2022-01-05:
Hi,

I have modified :

```
mp_float_t cosAngle = cosf(angle);
mp_float_t sinAngle = sinf(angle);
```
perfect ! It's works

I don't know if it closes

Thank you for your support