russhughes/st7789_mpy #55: build error for PYBV11

erongd, 2022-01-18:
I am trying to build the firmware for PYBV11. 

I build the firmware with the following command.
 
make USER_C_MODULES=~/st7789_mpy BOARD=PYBV11 -j2

Got the following error. Would appreciate any points on resolving this issue.

CC /home/erong/st7789_mpy/st7789/mpfile.c
/home/erong/st7789_mpy/st7789/st7789.c: In function 'st7789_ST7789_draw_len':
/home/erong/st7789_mpy/st7789/st7789.c:628:24: error: conversion to 'mp_int_t {aka int}' from 'mp_float_t {aka float}' may alter its value [-Werror=float-conversion]
  return mp_obj_new_int((int) print_width * scale);
                        ^
/home/erong/st7789_mpy/st7789/st7789.c: In function 'st7789_ST7789_make_new':
/home/erong/st7789_mpy/st7789/st7789.c:2147:25: error: passing argument 1 of 'mp_raise_ValueError' from incompatible pointer type [-Werror=incompatible-pointer-types]
     mp_raise_ValueError("rotations tuple must have 5 elements");
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/erong/st7789_mpy/st7789/st7789.c:38:0:
../../py/runtime.h:182:15: note: expected 'mp_rom_error_text_t {aka struct <anonymous> *}' but argument is of type 'char *'
 NORETURN void mp_raise_ValueError(mp_rom_error_text_t msg);
               ^~~~~~~~~~~~~~~~~~~
CC /home/erong/st7789_mpy/st7789/tjpgd565.c
CC ../../lib/oofatfs/ff.c
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:77: build-PYBV11/st7789/st7789.o] Error 1
make: *** Waiting for unfinished jobs....

russhughes, 2022-01-18:
I couldn't replicate the errors using my build system, so I could not verify that https://github.com/russhughes/st7789_mpy/commit/48391834af64a576ec5b80e75bc9caed6ef9f9e0 fixes the issue.

erongd, 2022-01-18:
I am using the v1.17 branch of the micropython to build the firmware using the out of the box board definition of PYBV11.

I did try to build use the fixes, unfortunately it still not working. However some different error this time.

CC /home/erong/st7789_mpy/st7789/st7789.c
CC /home/erong/st7789_mpy/st7789/mpfile.c
/home/erong/st7789_mpy/st7789/st7789.c: In function 'st7789_ST7789_draw_len':
/home/erong/st7789_mpy/st7789/st7789.c:628:24: error: expected expression before 'int'
  return mp_obj_new_int(int(print_width * scale));
                        ^~~
/home/erong/st7789_mpy/st7789/st7789.c: In function 'RotatePolygon':
/home/erong/st7789_mpy/st7789/st7789.c:1773:29: error: conversion to 'float' from 'mp_float_t {aka double}' may alter its value [-Werror=float-conversion]
  mp_float_t cosAngle = cosf(angle);
                             ^~~~~
/home/erong/st7789_mpy/st7789/st7789.c:1774:29: error: conversion to 'float' from 'mp_float_t {aka double}' may alter its value [-Werror=float-conversion]
  mp_float_t sinAngle = sinf(angle);
                             ^~~~~
/home/erong/st7789_mpy/st7789/st7789.c: In function 'st7789_ST7789_draw_len':
/home/erong/st7789_mpy/st7789/st7789.c:629:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:77: build-PYBV11/st7789/st7789.o] Error 1
make: *** Waiting for unfinished jobs....

erongd, 2022-01-18:
I changed MICROPY_FLOAT_IMPL to float, the error also changed.

CC /home/erong/st7789_mpy/st7789/st7789.c
CC /home/erong/st7789_mpy/st7789/tjpgd565.c
/home/erong/st7789_mpy/st7789/st7789.c: In function 'st7789_ST7789_draw_len':
/home/erong/st7789_mpy/st7789/st7789.c:628:24: error: expected expression before 'int'
  return mp_obj_new_int(int(print_width * scale));
                        ^~~
/home/erong/st7789_mpy/st7789/st7789.c:629:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
make: *** [../../py/mkrules.mk:77: build-PYBV11/st7789/st7789.o] Error 1
make: *** Waiting for unfinished jobs....

erongd, 2022-01-18:
The build succeed when I made following changes on line #628 in "st7789.c"

from 
return mp_obj_new_int(int(print_width * scale));
to 
return mp_obj_new_int((int) (print_width * scale));

russhughes, 2022-01-18:
That my mistake; It should have been mp_obj_new_int((int) (print_width * scale)); 

st7789.c:1773:29: error: conversion to 'float' from 'mp_float_t {aka double}' may alter its value [-Werror=float-conversion]
st7789.c:1774:29: error: conversion to 'float' from 'mp_float_t {aka double}' may alter its value [-Werror=float-conversion]

I need to create a macro so the correct function is called for the expected float or double.

russhughes, 2022-01-18:
I think this should cover both errors.

erongd, 2022-01-18:
I confirm the change fixed all the build error with MICROPY_FLOAT_IMPL set to either float or double.

closed 2022-01-18 by erongd