russhughes/st7789_mpy #57: rotation

mmx64, 2022-01-24:
Last 2 build for RP2 rotation does not work, i uploaded firmware from december and rotation works!

russhughes, 2022-01-25:
Can you provide more details?

mmx64, 2022-01-25:
Sure, i already had a working code using your firmware(installed in december)
tft = st7789.ST7789(SPI(0, baudrate=40000000, sck=Pin(18), mosi=Pin(19)),128,160,reset=Pin(17, Pin.OUT),cs=Pin(20, Pin.OUT),dc=Pin(16, Pin.OUT),backlight=Pin(5, Pin.OUT),color_order=st7789.BGR,
                        rotation=1)
tft.init()
tft.inversion_mode(False)
and i played around with _threading and i could not enter repl anymore, so i had to nuke flash and install new firmware. I downloaded last firmware from this repoository install it, everything works just display was rotated. I chnage
 rotation=(1,2,3,4) and nothing changes. After i tryed previously firmware and same result, so i try the december firmware and all works oki.

russhughes, 2022-01-25:
I forgot to add a default rotations table for that size display. You can work around the issue by specifying a custom table like this:

```
# (MADCTL Register Value, width, height, start_x, start_y)x

ORIENTATIONS = [
    (st7789.RGB,                          128, 160, 0, 0),
    (st7789.MADCTL_MX | st7789.MADCTL_MV, 160, 128, 0, 0),
    (st7789.MADCTL_MY | st7789.MADCTL_MX, 128, 160, 0, 0),
    (st7789.MADCTL_MY | st7789.MADCTL_MV, 160, 128, 0, 0),
]

tft = st7789.ST7789(
    SPI(0, baudrate=40000000, sck=Pin(18), mosi=Pin(19)), 
    128,
    160,
    reset=Pin(17, Pin.OUT),
    cs=Pin(20, Pin.OUT),
    dc=Pin(16, Pin.OUT),
    backlight=Pin(5, Pin.OUT),
    color_order=st7789.BGR,
    rotations=ORIENTATIONS,
    rotation=1)
```

closed 2022-02-09 by russhughes