russhughes/st7789_mpy #58: ESP32C3 display or decode jpg file caused panic'ed

Walkline80, 2022-01-24:
I used IDF v4.3.1 to compile firmware, and here's my test code:

```python
from machine import SPI, Pin
import st7789

spi = SPI(
    1,
    baudrate=40000000,
    polarity=1,
    sck=Pin(6),
    mosi=Pin(7)
)

display = st7789.ST7789(
    spi, 240, 240,
    reset=Pin(10, Pin.OUT),
    dc=Pin(4, Pin.OUT)
)

display.init()
display.jpg('images/luffy_60.jpg', 60, 60, st7789.SLOW)
```

Output:

```docs
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4200617e  RA      : 0x42004114  SP      : 0x3fcc3e20  GP      : 0x3fc9271c
TP      : 0x3fc90734  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0x0ff82f70  
S0/FP   : 0x3fcc3e50  S1      : 0x00000005  A0      : 0xd6067179  A1      : 0x3fcca7f0
A2      : 0x00000002  A3      : 0x00000c1c  A4      : 0x00000a1c  A5      : 0xd6067179  
A6      : 0x00000002  A7      : 0x00000008  S2      : 0x3fcca7f0  S3      : 0x00000003
S4      : 0x3c133000  S5      : 0x3fcc74f6  S6      : 0x3c164054  S7      : 0x00000038  
S8      : 0x0000004c  S9      : 0x0000000d  S10     : 0x3c13e000  S11     : 0x00000000
T3      : 0x00001000  T4      : 0x00000200  T5      : 0x00003010  T6      : 0x00002a3d  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0xd6067181
MHARTID : 0x00000000

Stack memory:
3fcc3e20: 0x3c164054 0x00000002 0x3fcca7f0 0xd6067179 0x3fcca610 0x3fccb528 0x00000001 0x42027fba
3fcc3e40: 0x00000076 0x3fcc53d0 0x3fcc3e80 0x42004114 0x3fcc9f78 0x00000002 0x3fcca7f0 0x3fcc3f04
3fcc3e60: 0x3fc9259c 0x3fcc53d0 0x3fcc3ea0 0x420040dc 0x3fcc9f78 0x00000005 0x3fcc3f04 0x4004a632
3fcc3e80: 0x3fcca610 0x00001a82 0xffffffff 0x3fcca730 0x3fcc9f78 0x00000005 0x3fcc3ed0 0x00000000
3fcc3ea0: 0x3c13e000 0x0000000d 0x0000004c 0x00000038 0x3c164054 0x3fcc74f6 0x3c133000 0x00000000
3fcc3ec0: 0x3fcc9f78 0x00000005 0x3fcc3fb0 0x420044a0 0x0000000c 0x00000000 0x3fcc9f78 0x00000005
3fcc3ee0: 0x4204a8f6 0x3fcca730 0x00000000 0x00000029 0x4038f372 0x42031560 0x400005a8 0x00000000
3fcc3f00: 0x3fc90734 0x4005890e 0x400005a8 0x3fcca7f0 0x00000000 0x00000004 0x00263513 0x00000049
3fcc3f20: 0x00000049 0x60023044 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcc3f40: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcc3f60: 0x00000000 0x00000000 0x000001ab 0x42016090 0x3fcca9f0 0x00000a1c 0x420040dc 0x3fcc3ee4
3fcc3f80: 0x3c133ee0 0x0000003c 0x0000003c 0x3fcc75c0 0x3fcca590 0x3fcc74e3 0x3c133000 0x00000001
3fcc3fa0: 0x000001ab 0x3c164054 0x3c1302c8 0x420147d0 0x3fffffff 0x3fcc9f88 0x3fcc74f4 0x4201bf64
3fcc3fc0: 0x3fffffff 0x000005d0 0x00000003 0x3fc9bbc8 0x3fffffff 0x3fcc9f74 0x3fcc74f5 0x40382456
3fcc3fe0: 0x00000000 0x3fcc9e84 0x3fcc9fac 0x3fcc9f60 0x3c139690 0x3fcc9fa8 0x4201a584 0x00000000
3fcc4000: 0x00000000 0x3fcc3f90 0x00000002 0x3fcc9fa0 0x00000020 0x00000080 0x3fcc41e8 0x3fcd5000
3fcc4020: 0x40382070 0x3fcc41b0 0x3fc9bc60 0x00000002 0x00000000 0x3fcc436c 0x0000000e 0x3fcc9f60
3fcc4040: 0x00000038 0x0000004c 0x0000000d 0x3c13e000 0x00000000 0x3fcc3fe0 0x00000000 0xffffffff
3fcc4060: 0x00000000 0x00000000 0x02625a00 0x00000080 0x00000000 0x00000000 0x00000000 0x00000001
3fcc4080: 0x0000028a 0x00000002 0x00000008 0x00000006 0x00000007 0x00000001 0x00000000 0x00000008
3fcc40a0: 0x00000000 0x00000102 0x3fc984d0 0x42035114 0x00000080 0x00000011 0x3fcc40c0 0x3c13f1b8
3fcc40c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcc9ac0 0x00000000
3fcc40e0: 0x3c13e000 0x0000000d 0x00000050 0x3fcc4350 0x3fcc7830 0x3fcc414c 0x3fffffff 0x4201a584
3fcc4100: 0x3c134b28 0x3c14ba38 0x3fcc9e20 0x42019982 0x3fcc7830 0x3fcc815a 0x3c133000 0x0000163a
3fcc4120: 0x3fcc4368 0x3fcc7d30 0x3fcc7d00 0x0000000e 0x00000038 0x00000002 0x0000000c 0x00000000
3fcc4140: 0x00000000 0x3fcc4368 0x3fcc9e80 0x00000000 0x3c13e000 0x0000000d 0x0000004c 0x00000038
3fcc4160: 0x3fcc9f60 0x0000000e 0x3fcc436c 0x00000000 0x00000002 0x3fc9bc60 0x3fcc41b0 0x4201470a
3fcc4180: 0x3fcc7830 0x3fcc8167 0x00000050 0x3fcc4350 0x3fcc7830 0x3fcc8174 0x3c133000 0x3fc9bbc8
3fcc41a0: 0x3fffffff 0x3fcc4368 0x3fcc8173 0x403824d8 0x00000000 0x3fcc7a64 0x3fcc4390 0x3fcc4350
3fcc41c0: 0x3c139690 0x3fcc438c 0x4201a584 0x00000000 0x3fcc7a60 0x3fcc4384 0x3fcc7a48 0x3fcc4384
3fcc41e0: 0x0000001f 0x3fcc787c 0x3fcc43f8 0x42040a02 0x40382070 0x3fcc43d0 0x3fc9bc60 0x00000000
3fcc4200: 0x00000000 0x00000000 0x0000000b 0x3fcc4350 0x00000000 0x00000050 0x0000000d 0x3c13e000
```

I also tested other functions, like `fill_circle()`, `blit_buffer()` and `text()`, there's no problem with my esp32c3

russhughes, 2022-01-24:
I have a couple of ESP32C3 devices that should arrive later this week for testing.

Walkline80, 2022-01-26:
> I have a couple of ESP32C3 devices that should arrive later this week for testing.

That's great, I'll waiting for your test result, and also I have other 3 questions about this lib

First is, if my esp32c3 board could draw an image on tft, is there any way to draw some text on the image with transparent background?

And then, I use `blit_buffer()` to draw some text to fill the whole screen, when I continued to fill second screen, I have to clean first one, but `fill()` or `fill_rect()` will caused a flash, is there any way to avoid it?

Last one is, `fill()` and `init()` method maybe missed in [Method](https://github.com/russhughes/st7789_mpy#methods) chapter of README?

Have  a nice  day! @russhughes

MG7789, 2022-01-27:
Hi,
tried building for esp32c3, but unfortunately unable launch even micropython. 
Is it possible to upload precompiled version for esp32c3? Or some instructions what should be modified to build for esp32c3?

russhughes, 2022-01-28:
I'm working on getting a stable C3 firmware running.

Walkline80, 2022-01-28:
@MG7789 could you build a pure esp32c3 firmware correctly?

[Here](https://github.com/micropython/micropython/issues/8204) is the method that how I fixed SPI fatal error, and then build firmware with st7789 lib by using command wrote in README page:

```bash
$ make USER_C_MODULES=../../../st7789_mpy/st7789/micropython.cmake
```

For now, all of the functions were correctly except `jpg()` :)

russhughes, 2022-01-29:
@Walkline80 Thanks for the SPI fix. 

The `mp_readinto` function in `mpfile.c` causes the crash.  This function calls the python `readinto` method to read from the JPG file. Not sure why yet.

Walkline80, 2022-01-29:
@russhughes According to this [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-guides/fatal-errors.html#instruction-access-fault-load-access-fault-store-access-fault), the `Load access fault` caused by:

```docs
This CPU exception happens when application attempts to execute, read from or write to an invalid memory location.
```

Combined with the `MTVAL` register  value in my Outpus, the value `0xd6067181` was not in the valid range(0x3fxxxxxx - 0x6xxxxxxx)

I hope this info could be useful, In addition, how can I recurrent this error in MicroPython code?

MG7789, 2022-01-31:
@russhughes  thank you for uploading ESP32c3 bin. Got it working. 
Good luck getting jpg() fixed!

Walkline80, 2022-02-05:
Hi, @russhughes

I'm curious if it is necessary to delay 500ms in [ST7789.init()](https://github.com/russhughes/st7789_mpy/blob/master/st7789/st7789.c#L1173) function? The whole `init()` process takes almost a second. I think this is too long. Is it possible to shorten this delay?

Also, I think `vscrdef()` and `vscsad()` methods were missed in README too, I found it because of this issue #59  that was closed yesterday.

russhughes, 2022-02-05:
Reading the datasheet and looking at other drivers 500ms is way too long, 120ms should work after the ST7789_DISPON command. 

I'll add the `vscrdef()` and `vscsad()` methods to the README.

russhughes, 2022-06-08:
@Walkline80 You can add the following lines to the Makefile in ports/es32 to add the idf.py monitor command.

```
monitor:
        idf.py $(IDFPY_FLAGS) -p $(PORT) -b $(BAUD) monitor
``` 

Use `make monitor` to start a serial console that will automatically decode core dumps for you. For example, if you build your MicroPython firmware with DEBUG=1, the core dump decoder will show you the source code line number for each stack frame in the core dump backtrace.

russhughes, 2022-06-08:
The fix for the crash was to increase the size of the freeRTOS idle task stack. I increased the idle stack from 1536 to 3072 by adding `CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=3072` to the beginning of the `ports/esp32/boards/sdkconfig.base` file. I did not test other values; smaller values may work as well.

closed 2022-06-08 by russhughes

Walkline80, 2022-06-11:
I have tried GENERIC_C3 firmware in this repo, and it's perfect to load and display a .jpg image. However, by adding `CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=3072` to my sdkconfig file, I still got the same panic'ed fatal...

My build environment:

1. MicroPython fetched with master
2. IDF version switched to v4.3.2
3. Added 2 user modules
4. Part of skdconfig params about STACK:
    a) CONFIG_ESP_SYSTEM_EVENT_TASK_STACK_SIZE=4096
    b) CONFIG_ESP_MAIN_TASK_STACK_SIZE=4096
    c) CONFIG_ESP_MINIMAL_SHARED_STACK_SIZE=4096
    d) CONFIG_FREERTOS_IDLE_TASK_STACKSIZE=3072

Is there any thing I made mistake?

BTW, I'm using Windwos WSL to build my firmware, so I cannot using `idf monitor` to tracing backtrace, too.

russhughes, 2022-06-11:
I had issues using v4.3.2 with the C3s. So I used ESP-IDF V4.4 to compile my firmware.

Walkline80, 2022-06-12:
Thanks @russhughes , it's works correctly now.

russhughes, 2022-06-12:
Excellent.