russhughes/st7789_mpy #60: T-Display RP2040 display not working

erongd, 2022-02-09:
I got my hands on the lilygo t-display rp2040 board. The firmware I built that works on pico does not work on t-display, after display initialization the backlight turned on. Appreciate any points on troubleshooting the issue.

russhughes, 2022-02-10:
I have not had any luck getting the RP2040 T-Display to work with anything other than the python based driver.

russhughes, 2022-02-11:
This should initialize the display:

```
from machine import Pin, SPI
import st7789

Pin(22, Pin.OUT, value=1)

spi = SPI(
    0,
    baudrate=40000000,
    polarity=1,
    phase=0,
    sck=Pin(2, Pin.OUT),
    mosi=Pin(3, Pin.OUT),
    miso=None)

tft = st7789.ST7789(
      spi,
      135,
      240,
      cs=Pin(5, Pin.OUT),
      dc=Pin(1, Pin.OUT),
      backlight=Pin(4, Pin.OUT))

tft.init()

```

erongd, 2022-02-11:
I will pull the latest code and try it out. Thanks for the quick update.

russhughes, 2022-02-12:
You can use baudrate=62500000 to give it a little more speed. Not sure why using Pin 0 for reset causes issues.