russhughes/st7789_mpy #62: Lolin-S2  (ESP32S2)

AndyThirtover, 2022-02-19:
Hi there.
  Would it be possible to build firmware for the ESP32S2? 
  My chip gives:

 Chip is ESP32-S2FH32
Features: WiFi, Embedded 4MB Flash, 105C temp rating, ADC and temperature sensor calibration in BLK2 of efuse
Crystal is 40MHz

   I tried flashing with the generic ESP32 firmware - but sadly this failed.

Double regards

Andy

russhughes, 2022-02-20:
Micropython has firmware (https://micropython.org/download/?mcu=esp32s2) for several ESP32-S2 based boards. If your board is one of the boards listed, it would be easy for me to compile a firmware with the driver.  If not, it will require a custom board definition. Do you have a link to information about the board?

AndyThirtover, 2022-02-21:
Hi there.   
  This is the best reference I can find:
    https://www.wemos.cc/en/latest/s2/s2_mini.html
  Because it is so small I believe it will make a great pairing with the 240x240 1.15in displays.

Double Regards

Andy

russhughes, 2022-02-23:
Try firmware/LOLIN_S2_MINI/firmware.bin from bfac0c3fd6ea82afa8acf4b16eaf0cafbf5eee7e.

AndyThirtover, 2022-02-24:
Here's the results of the first load:

MicroPython v1.18-141-gff9c70850-dirty on 2022-02-22; LOLIN_S2_MINI with ESP32-S2FN4R2
Type "help()" for more information.

I'm about to connect up a TFT screen.

AndyThirtover, 2022-02-24:
OK, so I have failed to get a screen working.   I'm using this schematic:
https://www.wemos.cc/en/latest/_static/files/sch_s2_mini_v1.0.0.pdf
and this tft_config:
`from machine import Pin, SPI
import st7789

TFA = 0
BFA = 0

def config(rotation=0, buffer_size=0, options=0):
    return st7789.ST7789(
        SPI(2, baudrate=31250000, sck=Pin(7), mosi=Pin(11)),
        240,
        240,
        reset=Pin(12, Pin.OUT),
        dc=Pin(13, Pin.OUT),
        backlight=Pin(14, Pin.OUT),
        rotation=rotation,
        options=options,
        buffer_size=buffer_size)
`
I also tried the options on pins 33,35.   There is less documentation available for the ESP32-S2 so I can't quite work out what the SPI number should be.  I've tried both 1 and 2.     I've been reading up about SPI Modes and I'm wondering if this is the issue.
Cheers

Andy

AndyThirtover, 2022-02-24:
And in other news.
 A 320x240 display with a CS line WORKS (Slap like etc).
Now I'll research how to work with a display that does not have CS.

AndyThirtover, 2022-02-28:
In extra more news.  The screen I'm using seems to requite a few more SPI params:

SPI(2, baudrate=30000000, polarity=1, phase=1, sck=Pin(7), mosi=Pin(11)),

I consider the firmware running and tests - Job Done Tick ;)

closed 2022-02-28 by AndyThirtover

russhughes, 2022-02-28:
Excellent.