russhughes/st7789_mpy #63: ImportError: no module named 'tft_config'

yaconsult, 2022-03-05:
I have a TTGO T-Display board that was recently purchased.  I am not
new to python but I am completely new to micropython and esp32.  The
demo program that comes flashed on the device worked as it should so
the hardware is good.

I flashed the following image using 'esptool --chip esp32 write_flash -z 0x1000 firmware.bin'
https://github.com/russhughes/st7789_mpy/blob/master/firmware/T-DISPLAY-ESP32/firmware.bin
esptool.py v3.1
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-D0WDQ6-V3 (revision 3)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:21:84:88:c6:dc
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x001a0fff...
Compressed 1702480 bytes to 1042136...
Wrote 1702480 bytes (1042136 compressed) at 0x00001000 in 91.8 seconds (effective 148.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

>>> help('modules')
NotoSansMono_32   greekp            uarray            usocket
NotoSans_32       greeks            uasyncio/__init__ ussl
NotoSerif_32      inconsolata_16    uasyncio/core     ustruct
__main__          inconsolata_32    uasyncio/event    usys
_boot             inisetup          uasyncio/funcs    utime
_onewire          italicc           uasyncio/lock     utimeq
_thread           italiccs          uasyncio/stream   uwebsocket
_uasyncio         marker            ubinascii         uzlib
_webrepl          math              ubluetooth        vga1_16x16
apa106            meteo             ucollections      vga1_16x32
axp202c           micropython       ucryptolib        vga1_8x16
btree             music             uctypes           vga1_8x8
builtins          neopixel          uerrno            vga1_bold_16x16
cmath             network           uhashlib          vga1_bold_16x32
df                ntptime           uheapq            vga2_16x16
dht               onewire           uio               vga2_8x16
ds18x20           romanc            ujson             vga2_8x8
esp               romancs           umachine          vga2_bold_16x16
esp32             romand            uos               vga2_bold_16x32
flashbdev         romanp            upip              webrepl
framebuf          romans            upip_utarfile     webrepl_setup
fs                scriptc           uplatform         websocket_helper
gc                scripts           urandom
greekc            st7789            ure
greekcs           symbol            uselect
Plus any modules on the filesystem
>>> 

I cannot get any of the example programs to work but I don't know what
else I might need to do besides flash the firmware.bin file.

I tried entering the st7789_mpy/examples/hello.py from the python
prompt after connecting with minicomm and this is the first error that
I get:

>>> import random
>>> import utime
>>> import st7789
>>> import tft_config
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'tft_config'
>>> import vga1_bold_16x32 as font
>>> 

Am I missing some libraries or something that I need to copy to the
T-Display?  Where is tft_config defined?  Or are the examples out of
sync with the code?

I didn't get that far, but I assume that I will need to copy the font files
and maybe some utilities to the board, or are they all frozen/included?
It looks like I can copy anything needed with ampy.

Thanks!

russhughes, 2022-03-05:
Most of the modules for the examples are frozen in the firmware but you need to copy the tft_config.py file from examples/configs/tdisplay_esp32/tft_config.py to your device.

yaconsult, 2022-03-06:
It works and is very fast writing to the display.  What a wonderful project you have here for experimenting and learning about micropython and the t-display board!  It's a great dev board that is much more fun than blinking a damned LED on the esp32 without a display!  LOL

How would a beginner know which files need to be copied to the device?  I presume that the fonts need to be copied to it for some of the other demo programs or are they frozen in the firmware?  Or is the config file the only one needed because it's the only one specific to the board?

Thanks again for a really great and educational repo!

russhughes, 2022-03-06:
You're welcome. I need to work on the documentation for the examples to clarify the tft_config.py files. I grew tired of editing the examples when testing different displays, so I moved the display config into a separate file for each device.

closed 2022-03-06 by russhughes