russhughes/st7789_mpy #64: Is it possible that the firmware for the T-Display ESP32 is corrupt?

AndyThirtover, 2022-03-06:
Hi Russ
    It seems that on download is this firmware is much smaller than the others (test on Mac and Linux).
  Here's an example:

-rw-rw-r-- 1 andy andy 1724608 Feb 19 18:42 ST7789-ESP32-1.18-firmware.bin
-rw-rw-r-- 1 andy andy  204823 Mar  6 18:07 ST7789-T-DISPLAY-1.18-firmware.bin

Regards

Andy

russhughes, 2022-03-06:
The size looks correct. The size can vary depending on the amount of additional frozen bytecode included in the firmware. Are you seeing issues with the firmware?

AndyThirtover, 2022-03-07:
Here's the issue that I get on both examples of the T-Display ESP32 that I have:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

I note that the generic ST7789 firmware does load an give a REPL etc.

Regards

Andy

AndyThirtover, 2022-03-07:
My bad!    Issues in download:

Working:
Entering REPL. Use Control-X to exit.
>
MicroPython v1.18-128-g2ea21abae-dirty on 2022-02-12; TTGO T-Display with ESP32
Type "help()" for more information.
>>>

closed 2022-03-07 by AndyThirtover