russhughes/st7789_mpy #68: Document vscrdef and vscsad

wesleywerner, 2022-04-13:
I really enjoy using this driver, but struggling to understand the usage of vscrdef and vscsad in the examples. 

- why are TFA and BFA (top free area, bottom free area) named so? How do the top/bottom values correlate to the scrolling mechanism? both these return 40 on my 135x240 display.
- What is the second (height) parameter given to `vscrdef()` for? I tried large and small values with no discernible effect
- can we scroll horizontally?

Thanks for a great driver, I am having a lot of fun with it so far.

russhughes, 2022-04-14:
You're welcome. I've updated the README with more information about scrolling. See https://github.com/russhughes/st7789_mpy#scrolling. It needs more work, but it explains why the TFA and BFA are set to 40 for 135x240 displays.

Only Vertical scrolling is supported. However, to get horizontal scrolling, you could rotate the orientation of the display.

wesleywerner, 2022-04-21:
Thank you, that is very helpful 👍

closed 2022-04-21 by wesleywerner