russhughes/st7789_mpy #71: font2bitmap create wrong MAP

erian747, 2022-05-24:
Nice library, working fine on my STM32 Blackpill

But i encountered a problem when i was trying to generate a larger truetype font
I ran: python3 ../../utils/font2bitmap.py NotoSansMono-Regular.ttf 42 -c 0x20-0x7f

I got:
--------------------------------------------------------------------------------------------
`# -*- coding: utf-8 -*-
# Converted from NotoSansMono-Regular.ttf using:
#     ../../utils/font2bitmap.py NotoSansMono-Regular.ttf 42 -c 0x20-0x7f

MAP = (
    'o4w1Z*},:'U&n8%R.m~Ejy`f+Y5vS[OA'
    '@{DIzW7a#_2ksKq<\"Jh0c);x$6t3!/'
    'i=lp>BH(G^bX]? LVMP|rNu-d9\\TeCF'
    'Qg'
)
BPP = 1
HEIGHT = 43
MAX_WIDTH = 25
_WIDTHS = \
--------------------------------------------------------------------------------------------
Seems like the MAP becomes unordered and it will not compile in Micropython

st7789_mpy git commit b4aa060b74e2d2490770fa92310571f7602059f9 
System: 5.13.0-41-generic #46~20.04.1-Ubuntu SMP Wed Apr 20 13:16:21 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
Python 3.8.10

russhughes, 2022-05-24:
I forgot to escape single quote characters in the MAP. Add a slash (\) before the embeded single quote (') character in the MAP.

Change:

MAP = (
'o4w1Z*},:'U&n8%R.m~Ejy`f+Y5vS[OA'
'@{DIzW7a#_2ksKq<"Jh0c);x$6t3!/'
'i=lp>BH(G^bX]? LVMP|rNu-d9\TeCF'
'Qg'
)

To:

MAP = (
'o4w1Z*},:\'U&n8%R.m~Ejy`f+Y5vS[OA'
'@{DIzW7a#_2ksKq<"Jh0c);x$6t3!/'
'i=lp>BH(G^bX]? LVMP|rNu-d9\TeCF'
'Qg'
)

And it should compile. I'll fix the font2bitmap.py script today.

erian747, 2022-05-25:
Thanks, that worked for that generated font
Seems like there are another case to that has to be covered (if not already included in fix for issue above), it is when a line in the MAP ends with backslash
Like line 2 in this map:
```text
MAP = (
    'tR2-UML!64E9S&nA[(Viw+a>{F$T^:e'
    'g rdHv@qcp,bWO'Km8j_`;Zs<=IQzoD\'
    '\?uJ~xNkGf/|\")1B}X*C]0#%Yy7.Phl'
    '53'
)
```
I tried escaping the last backslash with double backslash and it compiled fine
I also had to remove the backslash ? in line 3
Seems like the first backslash in line 3 was supposed to be placed at the end of line 2

russhughes, 2022-05-25:
I discovered two other issues; the current version should correctly handle the escapes for quotes, single quotes, and backslashes.
 
I'll look at adding support for additional escape sequences like hex and unicode.

closed 2022-06-08 by russhughes