russhughes/st7789_mpy #72: SoftSPI not working on LilyGo T-PicoC3

Poedersuiker, 2022-05-25:
I am trying to get the display on a T-PicoC3 [(link).](https://github.com/Xinyuan-LilyGO/T-PicoC3) From the factory the PicoC3 is programmed with Arduino and the display is working. But when I want to use MicroPython the screen stays blank.

My test code:
`from machine import Pin, SoftSPI, SPI
import st7789
from time import sleep
    
mosi_pin = Pin(3, mode = Pin.OUT)
miso_pin = Pin(14, mode = Pin.IN)
sck_pin =  Pin(2, mode = Pin.OUT)

BACKLIGHT_PIN = Pin(4,  Pin.OUT)
RESET_PIN = Pin(0, Pin.OUT)
DC_PIN = Pin(1, Pin.OUT)
CS_PIN = Pin(5,  Pin.OUT)

import vga1_bold_16x32 as font

spi = SoftSPI(baudrate=100000, polarity=1, phase=1, sck=sck_pin, mosi=mosi_pin, miso=miso_pin)
display = st7789.ST7789(
    spi, 135, 240,
    reset     = RESET_PIN,
    cs        = CS_PIN,
    dc        = DC_PIN,
    backlight = BACKLIGHT_PIN,
    rotation  = 1)
        
display.init()
display.fill(65535)
# draw white letters on a back background at 10 over and 20 down
display.text(font, "Hello World!", 10, 20, st7789.color565(255,255,255), st7789.color565(0,0,0))`

I saw the comment about the blank screen, but I cannot use the hardware SPI because of the hardwiring on the board. Does anyone have an Idea what is wrong?

russhughes, 2022-05-25:
Pin 22 is PWR_ON for the display. Set pin 22 to 1 to turn the display on.

Poedersuiker, 2022-05-26:
I added these lines to the code (started under the pin definition but also tried on the bottom).

`Power_PIN = Pin(22, Pin.OUT)
Power_PIN.value(1)`

But that didn't help.

russhughes, 2022-06-02:
I've got one on order. I'll take a look at it once it arrives.

ExperiMentor, 2022-06-06:
Isn't Power_PIN.value() active low? ie write 0 gives output high?
Try Power_PIN.value(0)

Poedersuiker, 2022-06-06:
The problem was resolved with the help of LilyGo. It seemed the configuration they gave was not completely correct.

https://github.com/Xinyuan-LilyGO/T-PicoC3/issues/4#issuecomment-1147037532

russhughes, 2022-06-22:
Working config module for LILYGO T-PicoC3 [examples/configs/t-picoc3/tft_config.py](https://github.com/russhughes/st7789_mpy/blob/ce578b168c19d78b16bad845daa30fa4f64238c5/examples/configs/t-picoc3/tft_config.py)

closed 2022-06-22 by russhughes