russhughes/st7789_mpy #73: PYBV11 firmware does not contains ST7789 driver just ST7687

yuran1978, 2022-05-29:
Hello. Upload firmware from PYB11 folder to pyboard v.1.1 when I run help('modules') it shows me ST7687
I tried to build it step by step for PYB11 but it ignoes USER_C_MODULES.
FYI. Firmware for RP2 shows correct library for st7789.

Output:
>>> help('modules')
__main__          micropython       uasyncio/lock     uos
_onewire          network           uasyncio/stream   uplatform
_uasyncio         onewire           ubinascii         urandom
builtins          pyb               ucollections      ure
cmath             **st7687**            uctypes           uselect
dht               stm               uerrno            usocket
framebuf          uarray            uhashlib          ustruct
gc                uasyncio/__init__ uheapq            usys
lcd160cr          uasyncio/core     uio               utime
lcd160cr_test     uasyncio/event    ujson             utimeq
math              uasyncio/funcs    umachine          uzlib

russhughes, 2022-05-30:
Yes, it looks like I included the wrong firmware image.

yuran1978, 2022-05-30:
Fixed. Awesome! Thank you for assistance! It will be ideal if some details about how to build project for PYBV11 will be added.

closed 2022-05-30 by yuran1978