russhughes/st7789_mpy #75: build error

hyansuper, 2022-06-02:
I use esp-idf v4.4 and latest micropython source. got the following error:
```<command-line>: error: expected identifier or '(' before numeric constant
/home/hyan/esp/mpy/micropython/ports/esp32/build-GENERIC/genhdr/moduledefs.h:63:56: note: in expansion of macro 'MODULE_ST7789_ENABLED'
 extern const struct _mp_obj_module_t mp_module_st7789, MODULE_ST7789_ENABLED;
                                                        ^~~~~~~~~~~~~~~~~~~~~
/home/hyan/esp/mpy/micropython/py/objmodule.c:169:5: error: macro "MP_ROM_PTR" passed 2 arguments, but takes just 1
     MICROPY_REGISTERED_MODULES
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/hyan/esp/mpy/micropython/py/objmodule.c:40:
/home/hyan/esp/mpy/micropython/ports/esp32/build-GENERIC/genhdr/moduledefs.h:65:66: error: 'MP_ROM_PTR' undeclared here (not in a function); did you mean 'MP_QSTR_PTR'?
 #define MODULE_DEF_MP_QSTR_ST7789 { MP_ROM_QSTR(MP_QSTR_st7789), MP_ROM_PTR(&mp_module_st7789, MODULE_ST7789_ENABLED) },
                                                                  ^~~~~~~~~~
/home/hyan/esp/mpy/micropython/ports/esp32/build-GENERIC/genhdr/moduledefs.h:179:5: note: in expansion of macro 'MODULE_DEF_MP_QSTR_ST7789'
     MODULE_DEF_MP_QSTR_ST7789 \
     ^~~~~~~~~~~~~~~~~~~~~~~~~
/home/hyan/esp/mpy/micropython/py/objmodule.c:169:5: note: in expansion of macro 'MICROPY_REGISTERED_MODULES'
     MICROPY_REGISTERED_MODULES
     ^~~~~~~~~~~~~~~~~~~~~~~~~~
[1233/1384] Building C object esp-idf/...mpy/micropython/py/objnamedtuple.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
Makefile:34: recipe for target 'all' failed
make: *** [all] Error 2
```

russhughes, 2022-06-02:
Looks like it's due to https://github.com/micropython/micropython/commit/efe23aca7154d21a9e8f283313c7a1ac29e05d86. I'll look at it tonight.

russhughes, 2022-06-02:
You should be able to change the last line in the st7789.c file from

`MP_REGISTER_MODULE(MP_QSTR_st7789, mp_module_st7789, MODULE_ST7789_ENABLED);`

to:

`MP_REGISTER_MODULE(MP_QSTR_st7789, mp_module_st7789)`

and it should compile.

hyansuper, 2022-06-03:
thank you！ very quick fix

closed 2022-06-03 by hyansuper