russhughes/st7789_mpy #76: no urequests in 'firmware\T-DISPLAY-ESP32 '

luomh126, 2022-06-17:
hi, i'm a newer for esp32 .i want build a software about wearther,and show the wearther data to screen display,it run on st7789.
i got the firmware about T-DISPLAY-ESP32 from your project. it is a great project! but i cannot use the 'urequests ' modules. what should i do. thanks!

luomh126, 2022-06-17:
i had followed the "Setup MicroPython Build Environmen" steps. but i got the err:
[image]

ExperiMentor, 2022-06-17:
Which board are you using? The ESP32 is not used to drive the display on T-PicoC3 (it's generally intended for WiFi, Bluetooth etc communications). The ready-built firmware for the RP2040 part of the board (in firmware/T-Display-RP2040) does include working drivers for the display.

luomh126, 2022-06-17:
hi thanks 

my board is esp-wroom-32, the scren display is SPI  ST7789V. and esp32 running firmware is T-DISPLAY-ESP32 (in firmware\T-DISPLAY-ESP32).

so does i choose wrong firmware? i should run RP2040 

> Which board are you using? The ESP32 is not used to drive the display on T-PicoC3 (it's generally intended for WiFi, Bluetooth etc communications). The ready-built firmware for the RP2040 part of the board (in firmware/T-Display-RP2040) does include working drivers for the display.

ExperiMentor, 2022-06-17:
No, you don't have an RP2040 processor on your board, so you cannot use that firmware.
I can't help you further - sorry!

luomh126, 2022-06-17:
> No, you don't have an RP2040 processor on your board, so you cannot use that firmware. I can't help you further - sorry!

thanks

russhughes, 2022-06-17:
The T-DISPLAY comes in ESP32, RP2040, and ESP32+RP2040 versions. The esp-wroom-32 should be able to use the firmware\T-DISPLAY-ESP32.  You will need to download the urequests.py module to the esp-wroom-32s flash in order to use it.  

The screenshot shows that the esp-idf is missing several required python packages. You need to run the esp-idf/install.sh script to install the missing packages.

closed 2022-07-15 by russhughes