russhughes/st7789_mpy #77: library Update for jpg

e135193, 2022-07-30:
I have updated library in jpg function. Now you can stream jpeg file diretly from buffer. In ESP32-CAM usage case, camera buffer can be stream direclt to the LCD. I added new function as jpgBuf().

It is tested and works

```
buf = camera.capture()
display.jpgBuf(buf, 0, 0,st7789.FAST)
```

mkuczynski11, 2022-10-13:
Hi, I was wondering if you could share your configuration? I have actually written nearly identical function as the one that you mentioned and I am trying to stream camera frames directly onto the LCD. The problem is that when trying to display 240x240 frames only 2 frames per second can be shown. I achieved around 25FPS with 96x96 config tho, but I would really like to display 240x240 JPEG pictures because my lcd has the exact size.  What performance do you achive? Anything would be helpful.

russhughes, 2022-10-15:
The jpg and jpg_decode methods now accept a buffer or a filename. See the alien.py example in examples/jpg.  I will test with an ESP-CAM board once I locate mine.

closed 2022-10-25 by russhughes