russhughes/st7789_mpy #81: Sorry for the dumb question - only the library files

feiticeir0, 2022-08-17:
Since I've never compiled anything related to micropython and I'm was not interested on having to compile it with any upgrade:

What If I'm only interested on the libraries (the py files) and not having them "embedded - for lack of a better word" on the firmware ? 
How can I do that ? 

Copying the py files to the root of the RPI and import them on the program . 

thank you !

ExperiMentor, 2022-08-18:
It's not clear what you are asking.

Given that micropython isn't compiled, you seem to be saying both that you don't want to have to edit the library *.py files, and then next thing that you do want to edit them.

Maybe an example task would help us understand?

feiticeir0, 2022-08-18:
Hi ! 
Probably I didn't explained myself well. I know the libraries need to be compiled, but can't I get them already compiled for micropython and just copy them to the Pico and used them ? regardless of the micropython version ? 
just copy the py files to the root of the PICO and import them in a program ?

russhughes, 2022-08-20:
You can copy the py files to the PICO and import them in a program regardless of the MicroPython version.  This requires more ram for the compile.  

You can also use the mpy-cross program to pre-compile the py files into a mpy file, upload the mpy to the PICO and import it into a program as long as the versions of MicroPython and mpy-cross match.  This saves memory since the file is already compiled.

feiticeir0, 2022-08-21:
Thank you and thank you for the explanation of the differences !  Will try both ways and see what's best for me ! But now I know how to compile Micropython ! :)

closed 2022-08-21 by feiticeir0