russhughes/st7789_mpy #82: Extra ┬ character printed on any character above 127

Typhoone, 2022-08-19:
Hi,

I'm trying to print some of the characters in the 256 Character Bit Mapped Fonts on a lilygo T-Display rp2040, specifically 176 ░ but any character above 127 gets prefixed with a ┬ so instead of ░ I get ┬░

All the characters from 0-254 all show up correctly, but once it gets to 128 the ┬ character gets put in as well.

I tried a modified version of the bitmap_fonts example using vga2 and the same thing happens there, is there some flag I need to turn on to support the wider range of characters properly?

I'm using the pre-packaged firmware, and have the config and font file next to main.py on the t-display.

Thanks for the help.

```import st7789
import tft_config
import utime
import vga2_8x8 as font

tft = tft_config.config(0)
char = chr(128)

def main():
    tft.init()
    tft.fill(st7789.BLACK)
    tft.text(
        font,
        char,
        tft.width() // 2 - 1 // 2 * font.WIDTH,
        (tft.height() // 2),
        st7789.WHITE,
        st7789.BLACK)
    utime.sleep(80)
    tft.fill(st7789.BLACK)

main()

russhughes, 2022-08-20:
Pass the char as an integer to print characters > 127.

Typhoone, 2022-08-20:
Ahh I see,
Also with a little help figured out I can build strings like this bytes([176,177,178])

Thanks heaps!

closed 2022-08-20 by Typhoone