russhughes/st7789_mpy #85: urequests module is not included into firmware.bin build

mantydze, 2022-09-09:
```
>>> help('modules')
NotoSansMono_32   greekcs           scripts           uplatform
NotoSans_32       greekp            sdcard            urandom
NotoSerif_32      greeks            st7789            ure
__main__          inconsolata_16    symbol            uselect
_boot             inconsolata_32    uarray            usocket
_onewire          inisetup          uasyncio/__init__ ussl
_thread           italicc           uasyncio/core     ustruct
_uasyncio         italiccs          uasyncio/event    usys
_webrepl          italict           uasyncio/funcs    utime
apa106            marker            uasyncio/lock     utimeq
axp202c           math              uasyncio/stream   uwebsocket
btree             meteo             ubinascii         uzlib
builtins          micropython       ubluetooth        vga1_16x16
cmath             music             ucollections      vga1_16x32
df                neopixel          ucryptolib        vga1_8x16
dht               network           uctypes           vga1_8x8
ds18x20           ntptime           uerrno            vga1_bold_16x16
esp               onewire           uhashlib          vga1_bold_16x32
esp32             romanc            uheapq            vga2_16x16
flashbdev         romancs           uio               vga2_8x16
focaltouch        romand            ujson             vga2_8x8
framebuf          romanp            umachine          vga2_bold_16x16
fs                romans            uos               vga2_bold_16x32
gc                romant            upip              webrepl
greekc            scriptc           upip_utarfile     webrepl_setup
Plus any modules on the filesystem
```

Can you include **urequests** module?
https://makeblock-micropython-api.readthedocs.io/en/latest/public_library/Third-party-libraries/urequests.html

hanung665, 2022-09-23:
Yes, please..
pyotp too if it possible.

russhughes, 2022-10-15:
The urequests module should be in now.  I'm not familiar with pyotp.

closed 2022-10-15 by russhughes