russhughes/st7789_mpy #86: Compile errors with ESP32s

robhamerling, 2022-09-27:
Since a few days I'm experiencing compile errors for ESP32 devices to which I want to add st7789_mpy. 
The errors are like:

`/home/rob/st7789_mpy/st7789/st7789.c: In function 'write_spi':
/home/rob/st7789_mpy/st7789/st7789.c:127:66: error: 'mp_obj_type_t' {aka 'const struct _mp_obj_type_t'} has no member named 'protocol'
  mp_machine_spi_p_t *spi_p = (mp_machine_spi_p_t *) spi_obj->type->protocol;
                                                                  ^~
`

Near the end of the listing a link appears: 
[https://github.com/micropython/micropython/wiki/Build-Troubleshooting](url)
The text tries to explain the cause of the errors, but I'm afraid I'm stuck. 

These types of errors appear only with ESP32-variants, not with TEENSY40 and Blackpill (stm32-F411CEU6).

russhughes, 2022-09-27:
I'll have to look into this.

robhamerling, 2022-09-28:
Sorry, I wasn't 100% correct. I have a script to generate firmware for several targets.
- I generate firmware for the blackpill, but without st7789 module.
- I tried to include st7789 for the TEENSY40 firmware, but somehow the include fails. This might be an issue with my script, I'll try to find out.

So you may forget the 'exceptions' in the last sentence of my initial report!

closed 2022-10-09 by russhughes