russhughes/st7789_mpy #87: unable to add st7789_mpy modules to PyBoard, Teensy40

robhamerling, 2022-09-30:
Generation of firmware with st7789_mpy modules fails with me. I use the following command for the pybv11, which seems to me according to docs.

`make DEBUG=0 BOARD=PYBV11 -j USER_C_MODULES=/home/rob/st7789_mpy/st7789/micropython.cmake`

And the compile succeeds (that is without error messages), but after loading this firmware to the PyBoard   'help("modules")'  doesn't show any st7789 modules...
I have the Micropython source at level 461 and the latest st7789_mpy rep.

russhughes, 2022-10-01:
I'm seeing the same issue as well. Something has changed. I'll look at it as soon as I solve the new ESP32 compile issue.

erongd, 2022-10-14:
stm32 port does not use cmake. Therefore a "micropython.mk" file is needed along side of "micropython.cmake".

The file should look like follows.

"micropython.mk"
```
ST7789_MOD_DIR := $(USERMOD_DIR)
SRC_USERMOD += $(addprefix $(ST7789_MOD_DIR)/, st7789.c)
SRC_USERMOD += $(addprefix $(ST7789_MOD_DIR)/, mpfile.c)
SRC_USERMOD += $(addprefix $(ST7789_MOD_DIR)/jpg/, tjpgd565.c)
SRC_USERMOD += $(addprefix $(ST7789_MOD_DIR)/png/, pngle.c)
SRC_USERMOD += $(addprefix $(ST7789_MOD_DIR)/png/, miniz.c)

CFLAGS_USERMOD += -I$(ST7789_MOD_DIR) -DMODULE_ST7789_ENABLED=1 -DMICROPY_PY_FILE_LIKE=1
CFLAGS_USERMOD += -DEXPOSE_EXTRA_METHODS=1
```

To build for PYBV11
```
make USER_C_MODULES=~/st7789_mpy BOARD=PYBV11 BOARD_VARIANT=dp-thread DEBUG=0 -j 2
```

commit cc905c81 (2022-10-14): Fix #87 PyBoard Build, #94 170x320 support
closed 2022-10-14 by russhughes