russhughes/st7789_mpy #89: twatch-2020-V2 : st7789 issue (and i2c too)

picpic020960, 2022-10-08:
Bonjour ,
I have twatch-2020-V1 and V2(GPS QUECTEL L76K ans SD card , not LORA)
st7789 with micropython works fine on V1 but on V2 'hello world' show only vertical lines with some color
the delta config V1/V2 for st7789 is BL 12/25
One idea of the reason ?
thank's for help

picpic020960, 2022-10-09:
OK with SoftSPI
thanks to #67

picpic020960, 2022-10-11:
I send 4 pixels in the middle of screen : 
V1 : OK for the pixels and background color
V2 : OK for the pixel , background color is bad , same as the above photo
how to debug ?
Thanks for help diagnostic

russhughes, 2022-10-15:
I have not had any problems with the twatch-2020 v2.  Make sure you are setting the display power to on with something like this:

```
import axp202c
axp = axp202c.PMU()
axp.enablePower(axp202c.AXP202_LDO2)
```

picpic020960, 2022-10-17:
Bonjour 
My V2 (with GPS and SD Card , **not LORA**) is 
**Chip is ESP32-D0WDQ6-V3 (revision v3.0)**
axp202c is good , I see my four pixels at screen center.
But the background color id bad ,as above photo .
seems to bee color format issue ...
some tricks ...

picpic020960, 2022-10-17:
"I have not had any problems with the twatch-2020 v2." : can you test i2c.scan() for the touch screen (sda=23 , scl=32).
 in my V2, empty list [] so error in Ftouch script.
Thanks for help

picpic020960, 2022-10-26:
[Résolvez] for I2C : for touche axp202 needs LDO2 , LDO3 and EXTEN
But for SPI  OK with lvgl with fliker

picpic020960, 2022-10-29:
OK all the trick in the TTGO.h , initpower : translate in python and all is good : no noise on screen and i2c.scan() reliable !

closed 2022-10-29 by picpic020960