russhughes/st7789_mpy #90: The clock  example program

yaconsult, 2022-10-10:
I am trying to run the clock example on a TTGO T-Display.   This is the older board, not the new TTGO T-Display S3 (which is on order).  I can run most of the other examples, but not the clock one.  I am using the firmware.bin from this git repo.

```
$ mpremote 
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] to exit this shell

MicroPython v1.19.1-292-g59e3348c1-dirty on 2022-09-04; TTGO T-Display with ESP32
Type "help()" for more information.
>>> import clock
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "clock.py", line 189, in <module>
  File "clock.py", line 118, in main
OSError: [Errno 2] ENOENT
>>> 

mpremote ls :
ls :
        6443 clock.py
        1338 hello.py
       11064 nasa01.jpg
        5507 nasa02.jpg
        2832 nasa03.jpg
        5089 nasa04.jpg
        6809 nasa05.jpg
       10537 nasa06.jpg
       10959 nasa07.jpg
        2421 nasa08.jpg
        9190 nasa09.jpg
        5615 nasa10.jpg
        8376 nasa11.jpg
        8485 nasa12.jpg
        4024 nasa13.jpg
        8205 nasa14.jpg
        5603 nasa15.jpg
        4943 nasa16.jpg
        7386 nasa17.jpg
        6190 nasa18.jpg
        5533 nasa19.jpg
        1588 nasa20.jpg
       12184 nasa21.jpg
       10042 nasa22.jpg
        9764 nasa23.jpg
       11885 nasa24.jpg
       18894 nasa25.jpg
        3962 pacifico40.py
       19076 st7789.pyi
         486 tft_config.py
       25141 vga1_bold_16x32.py`
```
These are all in the root folder of the t-display board.  The tft_config file is from:
st7789_mpy-master/examples/configs/tdisplay_esp32/tft_config.py

[https://gist.github.com/yaconsult/9200222f935a8f6313e353a4656f6d91](url)

Am I doing something wrong?  Does this clock example work for anyone else?

When I run the hello program with 'import hello' it runs and displays its output as expected.

Thanks!

russhughes, 2022-10-10:
Place the nasa*.jpg images in a folder named clock_240x135. 

Another solution is to remove line 117:

```
image_file = "clock_{}x{}/{}".format(tft.width(), tft.height(), image)

```

and change the line after from:

```
tft.jpg(image_file, 0, 0, st7789.SLOW)
```

to

```
tft.jpg(image, 0, 0, st7789.SLOW)

```

yaconsult, 2022-10-11:
I understand what you did now - it's written to work with all the different display resolutions based on which tft_config file is used.

Speaking of which, when the t-display S3 arrives, I assume the best approach will be to create a new tft_config for it since the display size increases from 135x240 to 170x320?  And that I will also need to generate new fonts since they are specific to the display size?  Oh, wait - I see that you already have a repo for the t-display S3!  Great!  I will start there.

Excuse the newbish question, but is there a way to have the driver in a loadable module or mpy file so that it could be used with any build of micropython?  Or is it necessary just to have a build environment and regenerate the firmware.bin whenever you want to incorporate an update to micropython and/or customize which modules/libraries are included?

Thank you for all your code and help!  This little board is a wonderful platform for being able to experiment with micropython without needing to mess with breadboards and wiring.

russhughes, 2022-10-11:
I test with many different boards and displays, so it made sense to do it this way; it should be better documented.  The loadable modules do not support hardware access, so the driver must be compiled along with MicroPython.  You have to rebuild it each time you want to update MicroPython.

You're welcome.

closed 2022-10-11 by russhughes