russhughes/st7789_mpy #98: T-Display RP2040: no display output

StevenCellist, 2022-10-22:
Hi guys, I feel like I am incapable of a basic achievement: getting any output on the ST7789 on my T-Display RP2040.

I have tried multiple examples, SPI configurations and Pin numbers, but the only thing that visually happens is the backlight turning on / off.
Zero errors when using the example, but just no output whatsoever. Calling Pin(4).value(0) and value(1) toggles the backlight but that's all I get.
Any suggestions or are there different revisions of the board?
Purchase link: https://www.tinytronics.nl/shop/en/development-boards/microcontroller-boards/others/lilygo-t-display-rp2040-with-1.14-inch-tft-display

russhughes, 2022-10-22:
I have one of these boards, I'll take a look at it later today.

closed 2022-10-24 by russhughes

russhughes, 2022-10-24:
Change the SPI named parameter `polarity=1` to `polarity=0` in the tft_config.py module, optionally add `reset=Pin(6, Pin.OUT)`

[tft_config.py](https://github.com/russhughes/st7789_mpy/blob/master/examples/configs/tdisplay_rp2040/tft_config.py)

commit 5bc5ea4b (2022-10-24): fix #98;  tft_config.py SPI polarity=0